arduino/arduino-ide #936: Opening example sketch does not check the .development file flag

KurtE, 2022-04-01:
### Describe the problem

If you open an example sketch and you have marked the library as being in development. 

As described in: https://arduino.github.io/arduino-cli/0.21/library-specification/#development-flag-file

The file is still opened up in the temporary directory not shown as within the library. 

On IDE1, the example is opened, I can edit it and then do a save and the example is updated in place. 

### To reproduce

mark a library as in development, by creating the .development file,
And open an example sketch within that folder.

### Expected behavior

Same as IDE1

### Arduino IDE version

running 0401 nightly

### Operating system

Windows

### Operating system version

10

### Additional context

I mentioned this in comment to #925 

@per1234 mentioned that the CLI does not even pickup the presences of this file:
https://arduino.github.io/arduino-cli/0.21/rpc/commands/#library

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

darianbjohnson, 2022-10-23:
I've got the same issue

fpistm, 2023-09-21:
Always present in 2.2.1