arduino/arduino-ide #951: Improve Linux installation instructions or update install.sh with udev rules

its-DomeE, 2022-04-11:
### Describe the request

The current installation instructions only mention to add the user to the dialout group to be able to access serial ports. On some (newer) distros this won't work anymore.

Instead you need to modify the udev rules. In my case I had to add: `SUBSYSTEMS=="usb-serial", TAG+="uaccess"` to `/etc/udev/rules.d/01-ttyusb.rules`.

It would be great if this information would be added to the install instructions or maybe even add this rule by `install.sh` itself.

### Describe the current behavior

If the current installation instructions are followed, serial ports can't be accessed on some distros because of a permission error.

### Arduino IDE version

1.8.19

### Operating system

Linux

### Operating system version

Zorin OS 16.1 (based on Ubuntu 20.04)

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

fstasi, 2022-04-13:
hi @its-DomeE thanks for opening this issue! We are very likely to accept a PR to improve our documentation or the install script. Let me know if you are interested contributing

its-DomeE, 2022-04-13:
Hi @fstasi, sure I'd love to contribute! I could look into this and create a PR after the Easter holidays.

matteocordray, 2022-04-16:
I'll agree to this, had to look through multiple websites just to get everything working/running on Ubuntu!