arduino/arduino-ide #964: File doesn't open if folder is has wrong name

Chose99, 2022-04-22:
### Describe the problem

If i want top open a file in a folder cold $filename$.ino the IDE opens but without the file and not error measge that it should be in a folder cold $filemane$

### To reproduce

Folder cold block2.ino with a block2.ino file in it

### Expected behavior

-the file just opens
- the file opens but you get the same error as in the IDE 1.8

### Arduino IDE version

2.0.0-rc6-nightly-20220422

### Operating system

Windows

### Operating system version

10.0.19044

### Additional context

Updating to the nightly build didn't change anything as far as i can tell no file opens if the folder isn't cold exactly like the file (without the extention)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-19:
>it does not work with valid sketches either.

Try it with the absolute sketch path @kittaakos. I get the same results from a relative path, but it works fine with the absolute sketch path.

kittaakos, 2022-10-19:
Thank you for verifying it on Windows. It does not work with absolute path either on macOS. I tried these:

```
% ls -al ./my_workspace 
total 0
drwxr-xr-x   3 a.kitta  staff   96 Oct 19 10:41 .
drwx------@ 16 a.kitta  staff  512 Oct 19 11:56 ..
-rw-r--r--   1 a.kitta  staff    0 Oct 19 10:41 my_workspace.ino
```

```
% /Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE /Users/a.kitta/Desktop/my_workspace 
```

```
% /Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE /Users/a.kitta/Desktop/my_workspace/my_workspace.ino
```

Electron [requires different file-association implementations](https://github.com/arduino/arduino-ide/blob/87ebcbe77e38d3ce435758c6836d9ca07f86acd0/arduino-ide-extension/src/electron-main/theia/electron-main-application.ts#L141-L148) for macOS and other OSs. I bet that is causing the problem in IDE2, but I do not understand why relative workspace files are broken on Windows. It should work. See [here](https://github.com/eclipse-theia/theia/blob/ca417a31e402bd35717d3314bf6254049d1dae44/packages/workspace/src/node/default-workspace-server.ts#L50-L53).

This is just a guess, but I think a good one. The `.ino` file association implementation overrules Theia's default CLI contribution:
 - on macOS, and that's why it does not work. The frontend does not have the chance to pick up the workspace path from the CLI, but IDE2 forcefully tries to restore all previously opened sketches.
 - on Windows and Linux, the default CLI contribution is also shadowed, but it's partially working with an absolute workspace root path.

per1234, 2022-10-19:
I checked on Linux and got the same result as on Windows: the sketch opens when I pass the absolute path via a command line argument, but it does not open when I pass the relative path.

Even with absolute path, it only works when I pass the path to the `.ino` file. It does not work when I pass the path to the sketch folder. Unfortunate, but this is also how Arduino IDE 1.x works so I think the users are accustomed to that quirk.

kittaakos, 2022-10-19:
> It does not work when I pass the path to the sketch folder.

I will fix this too as part of #1563

commit c0ca1869 (2022-10-18): fix: Prompt sketch move when opening an invalid outside from IDE2

Closes #964
commit 536e8990 (2022-10-18): fix: Prompt sketch move when opening an invalid outside from IDE2

Closes #964
commit 09b23369 (2022-10-18): fix: Prompt sketch move when opening an invalid outside from IDE2

Closes #964
commit 37c30d5e (2022-10-19): fix: Prompt sketch move when opening an invalid outside from IDE2

Log IDE2 version on start.

Closes #964
Closes #1484
closed 2022-10-26 by kittaakos