arduino/arduino-ide #969: Menu item "Edit > Select All" only select 10 lines of code

ubidefeo, 2022-04-25:
### Describe the problem

When wanting to select all the code in the current tab, if using the CMD+A shortcut it is correctly selected,
while using the menu item will only select 10 lines and it's not really clear where the starting point is, because the selection begin/end are offset from the last cursor position

### To reproduce

* open a Sketch
* place your cursor in the code window
* from the application menu __Edit > Select All__
* unselect by clicking in an unselected part of the text
* only 10 lines will be selected
* use the shortcut CMD+A (CTRL+A on Windows)
* the entire code will be selected

[image]

[image]

### Expected behavior

The entire code in the current tab should be selected and ready to be acted upon

### Arduino IDE version

2.0.0-rc6

### Operating system

macOS

### Operating system version

12.3

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-06-03:
PR: https://github.com/arduino/arduino-ide/pull/1022. Please review. Thank you!

kittaakos, 2022-06-08:
The PR is not merged. I have reopened this issue.

ubidefeo, 2022-06-22:
Tested PR and approved

commit dc65a781 (2022-06-08): Disabled `MenuItem` `roles` on macOS.

Closes #969
Upstream-ref: eclipse-theia/theia#11217
commit 0f8a29a4 (2022-06-22): Disabled `MenuItem` `roles` on macOS.

Closes #969
Upstream-ref: eclipse-theia/theia#11217
closed 2022-06-22 by kittaakos