arduino/arduino-ide #970: Renaming a tab does not remove the file from build cache

alranel, 2022-04-26:
### Describe the problem

When renaming a tab, the old file with its old contents gets still compiled during the build. This leads to unexpected and very confusing compilation results and errors. User has no way to fix the issue apart from locating the build cache directory and clean it up manually.

### To reproduce

1. Create a new sketch and save it.
2. Create a new tab (call it like `foo.c`) and put some non-compiling code in it (like `Foobar foo;`).
3. Use the "Rename" command to rename the tab to something else (like `bar.c`).
4. Replace the content of the tab with some legit code (like `#define FOOBAR`).
5. Compile the sketch.

You'll see an error related the bad code you actually deleted (`Foobar foo;`) located in a file which should not exist anymore (`foo.c`).
If you locate the temporary build cache directory, you'll see that the old file was not removed.

### Expected behavior

After a rename, the old file should be removed from the build directory and not be taken into account for compilation.

### Arduino IDE version

2.0.0-rc6

### Operating system

macOS

### Operating system version

10.15.7

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-04-26:
Thanks for taking the time to submit an issue @alranel .

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/1240.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

alranel, 2022-04-29:
Thank you @per1234. That's the upstream issue and it's good to link it but I recommend to keep this one open as well because it ends up being a bug of a feature of the IDE as well. If we close it here it can't be attached to any milestone in the IDE project, and this is a pretty severe bug that in my opinion blocks the release of IDE 2.0 stable, or at least should be evaluated for that milestone

per1234, 2022-11-15:
@davegarthsimpson same status as it has been for the last 6 years: unresolved.

In order to resolve it, the following things must be done: 

1. Fix https://github.com/arduino/arduino-cli/issues/1240
2. Create a new release of Arduino CLI
3. Bump Arduino IDE's Arduino CLI dependency to the version that contains the fix.

closed 2023-10-20 by per1234