arduino/arduino-ide #972: Serial Monitor autoscroll only makes bottom line partially visible

per1234, 2022-04-26:
### Describe the problem

The Arduino IDE "**Serial Monitor**" view has a "**Toggle Autoscroll**" control. When this is enabled, the view automatically scrolls down as the number of lines of data exceed the visible height of the view.

🐛 The autoscroll is not done completely, resulting in the most recently printed line only being partly visible.

### To reproduce

#### Equipment

- Any Arduino board capable of serial output.

#### Steps

1. Upload a sketch that produces sufficient serial output to exceed the height of the "**Serial Monitor**" view.
   For example:
   ```cpp
   int counter;
   void setup() {
     Serial.begin(9600);
   }
   void loop() {
     Serial.print("hello #");
     Serial.println(counter++);
     delay(500);
   }
   ```
1. Open the "**Serial Monitor**" view.
1. Select "**9600 baud**" from the dropdown baud rate menu at the top right corner of the "**Serial Monitor**" view.
1. Select the "**Toggle Autoscroll**" icon at the right side of the bottom panel toolbar if it is not already.
1. Wait for the serial output to exceed the visible height of the view.

🐛 The most recently printed line is only partly visible:

[image]

This is especially problematic when the line contains data that is important for the user to see at the time it is printed vs a constant data stream where the previous line serves just as well at the moment as the unreadable one.

### Expected behavior

The most recently printed line is always shown above the bottom edge of the Serial Monitor view.

### Arduino IDE version

2.0.0-rc6

### Operating system

Windows

### Operating system version

10

### Additional context

I bisected the issue to https://github.com/arduino/arduino-ide/commit/aba9db6a6b8b26c768ae6da09fabab58f382a9f1 (it does not occur when using the build from https://github.com/arduino/arduino-ide/commit/e5b34624ac297dd7f9d237342988848818c3b66b)

---

The line does become fully visible if I manually scroll the Serial Monitor down.

---

Originally reported at:

https://forum.arduino.cc/t/ide-2-0-rc6-serial-monitor-window-clipping-on-bottom-line/984677

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

mklemarczyk, 2022-06-24:
Same here, for version rc8 on Windows 10.

asier70, 2022-06-24:
It's very annoying to use the mouse scroll all the time to see last text line showed in Serial Monitor..
Windows 10, RC7.

KurtE, 2022-07-02:
Was double checking that issue was reported.   Still there on daily 0701

romainreignier, 2022-07-07:
The issue is also present on Linux (Ubuntu 18.04) rc8 AppImage.

asier70, 2022-07-19:
I noticed another problem conneted to this issue.
Trying to select text using mouse works not as expected.
Moving mouse pointer to the top part of text allows to select text in current line. Moving it to middle or lower horizontal position selects text in line below!
It looks like text is divided in the same way like it hides on the bottom of Serial monitor window.
Probably text displayed in Serial Monitor window is not displayed like classical text in Output window but drawed as graphics. Maybe some calculations of this graphic chars position is done wrong.

Defragster, 2022-07-24:
Confirm RC9 on Win10 still exhibits this clipping of the newest trailing line in SerMon output area.
Also note if 'select to end' with mouse - as the newest line is added it is also selected (as expected) : However when the added line is selected the autoscroll Does Fully Show the bottom line in the highlighted text.

Defragster, 2022-07-27:
Interesting screen display! Serial output was a static print waiting for user input. But the MATH was such that that last line was CLIPPED and then recalculated and redrawn repeatedly.
When it was in this state on Win 10 laptop a 2 finger scroll up/down would go clear and when full bottom would repeat the blurred redraw:
[image]

asier70, 2022-08-11:
rc9.2 - issue still exists.

pturri83, 2022-08-17:
This issue appears also on MacOS 12.5.

alec03128, 2022-09-11:
rc9.4 - issue still exists on Windows 10.
Changing font sizes and interface scaling does not solve the issue.

tahunus, 2022-09-15:
Issue still present in stable release 2.0.0 (Windows 11)

(In any case, KUDOS to the team for this release!! )

el-samiyel, 2022-09-15:
Issue still present in 2.0 release. Window 10.

And yes, KUDOS for producing the new IDE, its spectacular.

closed 2022-09-20 by AlbyIanna

per1234, 2022-09-27:
Resolved by https://github.com/arduino/arduino-ide/pull/1446

heliophagus, 2022-12-20:
The bad serial monitor scroll bug persists in the latest version I'm using:

Version: 2.0.3
Date: 2022-12-05T09:30:25.331Z
CLI Version: 0.29.0 [76251df9]

It's SUPER annoying because the output I need to pay attention to (a response to a query) only occurs on the last line of serial output, & I have to scroll down to see that line (which is otherwise completely hidden) every. single. time.

In fact I wasted several hours when I first encountered this bug because it seemed the query wasn't being responded to. But it was. Just hidden.

The puzzling thing is that this bug has been asserted as fixed for months now. But it isn't fixed. Will it ever be fixed? Can it, in fact, be fixed? Is this Kurt Gödel incompleteness territory? Strange times indeed.