arduino/arduino-ide #973: Auto Format (Ctrl+T) not working on Ubuntu 22.04 & IDE Version 2.0.0-rc6

FastThenLeft, 2022-04-29:
### Describe the problem

Auto Format (Ctrl+T) is not working on Ubuntu 22.04 & IDE Version 2.0.0-rc6.

### To reproduce

Write some unformatted code and press Ctrl+T or use Tools->Auto Format. Nothing happens.

### Expected behavior

The code should be formatted like it was in IDE 1.

### Arduino IDE version

2.0.0-rc6

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Additional context

Is there a package I need to install perhaps?

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

FastThenLeft, 2022-04-29:
I figured out what is happening. It's still a defect, but not as big.

If you do not select which board you are compiling for *before* you try to auto-format, then it does not work. If you select the board, then auto-formatting works.

per1234, 2022-06-08:
Thanks so much for your valuable report @FastThenLeft, and especially for taking the time to share the update with your later findings. I'm happy to tell you that this has been fixed by https://github.com/arduino/arduino-ide/pull/1019

If you would like to give it a try now, it is in the latest nightly build. The download links for the nightly build are listed here:

https://github.com/arduino/arduino-ide#nightly-builds

Otherwise, it will be in the next release of Arduino IDE 2.x