arduino/arduino-ide #975: No libraries

sonofcy, 2022-05-02:
### Describe the problem

The libraries pane is totally empty

### To reproduce

Click libraries

### Expected behavior

A list of libraries

### Arduino IDE version

2.0 rc6

### Operating system

macOS

### Operating system version

12.3.1

### Additional context

I have a substantial library. It shows up in IDE 1.0, and used to show up in IDE 2.0 rc5 I think but it might have been rc4

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-05-02:
Hi @sonofcy. Thanks for your report.

In order to gather more information that might help us to troubleshoot your problem, I'm going to ask you to run the Arduino IDE from the command line, then share the output.

Please do this:

1. Open your **Applications** folder.
1. Open **Utilities**.
1. Double-click on "**Terminal**".
1. Type the following command:
   ```text
   "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
   ```
1. Press **<kbd>Enter</kbd>**
1. Wait for the Arduino IDE to finish starting.
1. Switch back to the terminal window.
1. Press **<kbd>⌘</kbd>+<kbd>A</kbd>**.
   This will select all the text in the terminal window.
1. Press **<kbd>⌘</kbd>+<kbd>C</kbd>**.
   This will [copy](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the selected text to the clipboard.
1. Click on the "**Leave a comment**" field here on this GitHub issue.
1. Press **<kbd>⌘</kbd>+<kbd>V</kbd>**.
   This will [paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the output into the code block.
1. Paste the copied output into a reply here by pressing **<kbd>⌘</kbd>+<kbd>V</kbd>**.
   This will [paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the output into the code block.
1. Click the <kbd>**Comment**</kbd> button to make the reply.

---

Please let me know if you have any questions or problems while following those instructions.

sonofcy, 2022-05-02:
Last login: Tue Apr 26 14:17:51 on ttys000

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
Rons-MacBook-Air:~ ronalexander$ ping RonsHost
ping: cannot resolve RonsHost: Unknown host
Rons-MacBook-Air:~ ronalexander$ /Applications/Arduino\ IDE.app 
-bash: /Applications/Arduino IDE.app: is a directory
Rons-MacBook-Air:~ ronalexander$ "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE" 
Starting backend process. PID: 40034
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///Users/ronalexander/.theia'
root INFO Theia app listening on http://localhost:63108.
root WARN The local plugin referenced by local-dir:/Users/ronalexander/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/Users/ronalexander/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:/Users/ronalexander/.arduinoProIDE/plugins does not exist.
root WARN Backend LocalizationBackendContribution.initialize is slow, took: 65.0 ms
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-nl" to a VS Code extension "vscode-language-pack-nl@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-tr" to a VS Code extension "vscode-language-pack-tr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-zh-hans" to a VS Code extension "vscode-language-pack-zh-hans@1.53.2" with engines:
root WARN Backend DefaultWorkspaceServer.onStart is slow, took: 93.8 ms
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/cortex-debug/extension/dist/extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.2-beta.2" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-arduino-tools/extension/dist/bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-builtin-cpp/extension"
root INFO Deploying backend plugin "json@1.46.1" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-builtin-json/extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-builtin-json-language-features/extension/client/dist/jsonMain"
root INFO Deploying backend plugin "vscode-language-pack-de@1.53.2" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-language-pack-de/extension"
root INFO Deploying backend plugin "vscode-language-pack-fr@1.53.2" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-language-pack-fr/extension"
daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
root WARN Backend ConfigServiceImpl.onStart is slow, took: 399.7 ms
root INFO Deploying backend plugin "vscode-language-pack-ja@1.53.2" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-language-pack-ja/extension"
root INFO Refreshing token auth0|562ff5be-2829-4d08-842a-eb2506082861
daemon INFO Using config file: /Users/ronalexander/.arduinoIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.21.0
daemon INFO Executing `arduino-cli daemon`
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO {"IP":"127.0.0.1","Port":"63111"}
daemon INFO Daemon is running.
root WARN Backend AuthenticationServiceImpl.onStart is slow, took: 209.0 ms
root INFO Deploying backend plugin "vscode-language-pack-nl@1.48.3" from "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/vscode-language-pack-nl/extension"
[... cut ...]
daemon INFO Adding libraries dir [dir: /Users/ronalexander/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries, location: platform]
daemon INFO Adding libraries dir [dir: /Users/ronalexander/Documents/Source Code/Arduino/libraries, location: user]
daemon INFO Resolving include InqPortal.h for arch esp8266
daemon INFO   found better lib [lib: InqPortal, prio: 64D]
daemon INFO Resolving include Hash.h for arch esp8266
daemon INFO   found better lib [lib: AsyncElegantOTA, prio: 3F5]
daemon INFO   found better lib [lib: Hash, prio: 64C]

sonofcy, 2022-05-02:
Is instruction 12 a duplicate of 11, I only pasted the data one time in the leave a comment and then jumped to step 13.
This is my first use of github, at 80 I am still learning LOL.

per1234, 2022-05-02:
Thanks! I can see the problem from this part of the output:

>```text
>daemon INFO ERROR: 2022/05/02 12:08:52 [core] grpc: server failed to encode response:  rpc error: code = Internal desc = grpc: error while marshaling: string field contains invalid UTF-8
>root ERROR Request installed failed with error: 13 INTERNAL: grpc: error while marshaling: string field contains invalid UTF-8
>```

This is a known deficiency in the Arduino IDE's handling of character encodings used in the text of [the `library.properties` metadata file](https://arduino.github.io/arduino-cli/0.21/library-specification/#library-metadata): https://github.com/arduino/arduino-ide/issues/135

If you have any library installed which uses such encoding in its metadata, it breaks the library capabilities of the Arduino IDE.

---

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

---

The workaround is to adjust the encoding of the file. Unfortunately, the error messages do not indicate which library(s) have the problematic metadata, which makes finding the file that requires adjustment a bit inconvenient. If you would like assistance with that process, I would be happy to help you out over on the Arduino Forum, just make a post in the dedicated section for Arduino IDE 2.x support:

https://forum.arduino.cc/c/software/arduino-ide-2-0-beta/93

---

>Is instruction 12 a duplicate of 11

Yes. That was an error on my part. I apologize for any confusion. You posted the output perfectly.

>This is my first use of github

Welcome!

closed 2022-05-02 by per1234