arduino/arduino-ide #976: ERROR: "Unable to write into user settings..." when scaling UI using keyboard shortcuts

ubidefeo, 2022-05-03:
### Describe the problem

Changing the UI scaling via the Preferences panel works well,
but users who resort to shortcuts to zoom in/out the UI and font sometimes repeat the CMD/CTRL +/- to achieve the desired size.

When performing such shortcuts too rapidly the first action is not yet committed to the preferences file and the second one fails as the file has unsaved changes

### To reproduce

* Launch the IDE
* rapidly press CTRL/CMD +/- to scale the interface a few times

see video

https://user-images.githubusercontent.com/1894487/166421616-40b0a180-808a-45a7-b25a-f620a36cbb44.mp4

### Expected behavior

Theoretically changes should be either queued or buffered before being committed to the preferences file,
Canceling any shortcut action until the file is saved might turn into a frustration moment.
I often hit the combo 3 times while teaching or doing webinars and I do want the 3 degrees of zoom being applied.
If I had to give it time to save it might take me 9 presses and it would look like 6 were ignored for unknown reasons

### Arduino IDE version

2.0.0-rc6

### Operating system

macOS

### Operating system version

12.3

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-12-14:
I think the underlying issue was fixed here: https://github.com/arduino/arduino-ide/pull/856. Then it was also fixed in Theia, and IDE2 already uses it. Also, there was a significant improvement with event debouncing here https://github.com/arduino/arduino-ide/pull/1502/, and another fix to avoid saving the scaling twice here https://github.com/arduino/arduino-ide/commit/81195431b00b458d852fb4cd1c794a2596744b93.

Is this issue still valid? I have not noticed any issues when scaling the interface from the keyboard. Thank you!

per1234, 2023-09-28:
Since there hasn't been any indication that users are still experiencing this fault, I'll close the issue as resolved. If anyone is still experiencing it with the latest version Arduino IDE, feel free to comment here and we'll investigate.

closed 2023-09-28 by per1234