arduino/arduino-ide #980: NRF24L01 and Raspberry Pi Pico on Arduino IDE RC6

tootzreyes, 2022-05-07:
### Describe the problem

The NRF24L01+ connected to Raspberry Pi PIco does not work properly in Arduino IDE 2.0 RC6 while its working properly on Arduino IDE 2.0 RC5

### To reproduce

Here are the steps to replicate the issue:
1. Setup NRF24L01+ and Raspberry PI Pico using the default SPI pins and use the NRF24L01+ samples on Arduino using Arduino IDE 2.0 RC5. 
2. Upgrade the Arduino IDE 2.0 to RC6
3. Test the same code used in step 1. You would see that the radio initialization does not work

### Expected behavior

NRF24L01+ on Raspberry Pi Pico and Arduino IDE 2.0 RC5 and above should still have the radio initialization work

### Arduino IDE version

Arduino IDE 2.0 RC6

### Operating system

macOS

### Operating system version

MacOs Big Sur 11.6.5

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-05-07:
Hi @tootzreyes. Thanks for your interest in this open source project. Due to the limited information, the initial support for this problem will be better done via the Arduino Forum. I'm sure we will be able to help you out over there. If you haven't done so already, please make a post on the forum and I'll be sure to take a look:

https://forum.arduino.cc/

If it does turn out that the problem is caused by a previously unknown problem with the Arduino IDE 2.x code base during the course of the discussion, I'll be happy to reopen this.

closed 2022-05-07 by per1234

tootzreyes, 2022-05-08:
Please reopen.
I have clearly mentioned the details and the steps to replicate and its not out of topic as well.