arduino/arduino-ide #981: CPU usage too high

JachinY, 2022-05-07:
### Describe the problem

CPU usage too high
[image]

### To reproduce

[image]

### Expected behavior

[image]

### Arduino IDE version

2.0.0-rc6

### Operating system

Windows

### Operating system version

win11

### Additional context

[image]

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2022-05-07:
@JachinY 
This is a very unfriendly way of reporting an issue and your should either compile a proper one or retract it.
I do not tolerate such behaviour.

Thank you

JachinY, 2022-05-08:
Ok sorry thank you I dont know english

closed 2022-05-08 by JachinY