arduino/arduino-ide #984: Arduino IDE 2.0.0-rc6 (Linux AppImage 64-bit) does not start on Ubuntu 22.04

gmacario, 2022-05-11:
### Describe the problem

The Linux AppImage version assumes that libfuse.so.2 is installed, but I could not find it documented.

### To reproduce

1. Open <https://www.arduino.cc/en/software>
2. Select Arduino IDE 2.0 RC (2.0.0-rc6) : Linux AppImage 64 bit (X86-64)
3. When the download is complete, launch a Terminal and type
   ```bash
   cd ~/Downloads
   chmod 755 arduino-ide_2.0.0-rc6_Linux_64bit.AppImage
   ./arduino-ide_2.0.0-rc6_Linux_64bit.AppImage
   ```

Result:

```text
gmacario@hw2228:~/Downloads$ ls -la arduino-ide_2.0.0-rc6_Linux_64bit.AppImage 
-rwxr-xr-x 1 gmacario gmacario 241520658 mag 11 07:51 arduino-ide_2.0.0-rc6_Linux_64bit.AppImage
gmacario@hw2228:~/Downloads$ ./arduino-ide_2.0.0-rc6_Linux_64bit.AppImage 
dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
gmacario@hw2228:~/Downloads$
```

### Expected behavior

The Arduino IDE should launch correctly

### Arduino IDE version

2.0.0-rc6

### Operating system

Linux

### Operating system version

Ubuntu 22.04 LTS 64-bit

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

gmacario, 2022-05-11:
The workaround is pretty straightforward:

```bash
sudo apt install libfuse2
```

Result:

```ŧext
gmacario@hw2228:~/Downloads$ sudo apt install libfuse2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  libfuse2
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 90,3 kB of archives.
After this operation, 330 kB of additional disk space will be used.
Get:1 http://it.archive.ubuntu.com/ubuntu jammy/universe amd64 libfuse2 amd64 2.9.9-5ubuntu3 [90,3 kB]
Fetched 90,3 kB in 0s (644 kB/s)    
Selecting previously unselected package libfuse2:amd64.
(Reading database ... 204825 files and directories currently installed.)
Preparing to unpack .../libfuse2_2.9.9-5ubuntu3_amd64.deb ...
Unpacking libfuse2:amd64 (2.9.9-5ubuntu3) ...
Setting up libfuse2:amd64 (2.9.9-5ubuntu3) ...
Processing triggers for libc-bin (2.35-0ubuntu3) ...
gmacario@hw2228:~/Downloads$ ./arduino-ide_2.0.0-rc6_Linux_64bit.AppImage
```

then the "Arduino IDE - Terms of Service" dialog box is displayed and all proceeds as expected.

gmacario, 2022-05-11:
Where does the Arduino maintainters would suggest to document the dependency on `libfuse2` and how to install this library on the major Operating Systems?

probonopd, 2022-05-13:
We are experimenting with a statically linked AppImage runtime that won't require libfuse2 to be installed.

* https://github.com/AppImage/AppImageKit/issues/877

per1234, 2022-05-15:
Thanks for bringing this issue to our attention @gmacario and for sharing the solution you found.

Thanks also to @probonopd for the valuable information about the related upstream work.

I suspect that it might be some time before we are able to start distributing Arduino IDE AppImage packages which do not have this dependency. So I do think that it will be a good idea to document the fix.

Since I think the average Arduino user will be more likely to encounter this error and then search for a solution to the problem, rather than installing the potential dependency proactively after carefully following an installation guide, I propose that it be documented in a new article under the "Arduino Help Center".

Source here:

https://github.com/arduino/help-center-content/tree/main/content/Software%20and%20Downloads/IDE2.0

Published here:

https://support.arduino.cc/hc/en-us/sections/360005486620-IDE2-0

I propose that a note also be added to the installation instructions, linking to the Help Center article.

Source here:

https://github.com/arduino/docs-content/blob/main/content/software/ide-v2/tutorials/getting-started/01.ide-v2-downloading-and-installing/ide-v2-downloading-and-installing.md#linux

Published here:

https://docs.arduino.cc/software/ide-v2/tutorials/getting-started/ide-v2-downloading-and-installing#linux

KurtE, 2022-06-08:
Just ran into this today installing daily build 0608 to see if I had same issue on Ubuntu as I appear to see on windows, where no boards or libraries show up...

KurtE, 2022-09-28:
I thought I would mention, that I just had to reinstall Ubuntu on my test machine... 
Some how the sudo apt upgrade... gave some cryptic message about the multi-boot, and need to give password, and then will need it again after...  So then it rebooted.  And then I ended up with Panic errors... Could not read sector 0 or some such thing...

Ended up booting off of USB Stick and reinstall...

So then fresh install, downloaded the appimage, for the released IDE 2, and 
Followed the instructions, to set the properties to run... Double click does not work.

Run from Terminal window and it failed to run dlopen(): error loading libfuse.so.2
...

Should probably do something about this, at least up at:
https://www.arduino.cc/en/software

probonopd, 2022-09-29:
`sudo apt-get -y install libfuse2`.

KurtE, 2022-09-29:
Thanks.

I fixed it I believe installing libfuse-dev 

Yesterdays - comment was more along the line of it would be good if Arduino made some mention of this on their install page.
And likewise on the instructions page: 
https://docs.arduino.cc/software/ide-v2/tutorials/getting-started/ide-v2-downloading-and-installing

davegarthsimpson, 2022-11-14:
@per1234 can we close this one?

per1234, 2022-11-15:
The issue is not resolved, so I don't think it is appropriate to close it.

The problem can't be resolved in this repository at this time so that leaves the alternative I proposed of documenting the solution in [the Arduino Help Center](https://support.arduino.cc/hc/en-us). I don't have the permissions to transfer the issue to https://github.com/arduino/help-center-content (it requires write permissions in both repositories), but I'll put it on my "to do" list to submit a PR to the Help Center repository to document this. It will be a trivial task because it can simply consist of a link to the existing documentation on the subject:

https://docs.appimage.org/user-guide/troubleshooting/fuse.html

seaxwi, 2023-04-03:
@per1234 I've created a PR addressing this issue as it pertains to Help Center content:

https://github.com/arduino/help-center-content/pull/196