arduino/arduino-ide #989: Keyboard bindings do not work properly after change to different keyboard layouts

Azurac, 2022-05-14:
### Describe the problem

I have a keyboard layout where the letter Y is swapped with Z (Czech).
When I try to do Undo operation I need to press Ctrl + Y (Z in English layout) instead of Ctrl + Z (which is presented in the key bindings)

### To reproduce

Set a different keyboard layout in your system.
Start IDE with your default keyboard layout (e.g. EN)
Change it to different layout (e.g. Czech)

### Expected behavior

Set a different keyboard layout in your system.
Start IDE with your default keyboard layout (e.g. EN)
Verify you are typing right characters.
Verify Ctrl + Z and Ctrl + Y works as expected.
Change it to different layout (e.g. Czech)
Verify you are typing right characters.
Now Ctrl + Z and Ctrl + Y will work with the same buttons as previously, no matter the keyboard layout is.

### Arduino IDE version

2.0.0-rc6-nightly-20220514

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-09-21:
Upstream problem https://github.com/eclipse-theia/theia/issues/11688

closed 2022-09-26 by kittaakos

samtd, 2022-11-25:
I check with version 2.0.2 and the problem persist. The key number one don't work.
Portuguese Keyboard

kittaakos, 2022-11-27:
> I check with version 2.0.2 and the problem persist

OS?

> The key number one don't work.

Please explain what's not working.

> Portuguese Keyboard

[Here](https://github.com/eclipse-theia/theia/tree/master/packages/core/src/common/keyboard/layouts) are the supported keyboard layouts in Theia. The Portuguese ones:
 - [pt-Brazilian-mac.json](https://github.com/eclipse-theia/theia/blob/master/packages/core/src/common/keyboard/layouts/pt-Brazilian-mac.json)

 - [pt-Portuguese-mac.json](https://github.com/eclipse-theia/theia/blob/master/packages/core/src/common/keyboard/layouts/pt-Portuguese-mac.json)

 - [pt-Portuguese-pc.json](https://github.com/eclipse-theia/theia/blob/master/packages/core/src/common/keyboard/layouts/pt-Portuguese-pc.json)

If you are using a Brazilian Portuguese keyboard layout on Windows, we know why it's not working on your side.

samtd, 2022-12-07:
> 

Hi kittakos

I'm using Portuguese pt pc keyboard.

Also i already test on version v2.0.3 and the problem maintains.
With Arduino 1x i don't have any problem.

kittaakos, 2022-12-08:
@samtd, please help us to narrow down the problem. IDE2 builds on the [Theia IDE](https://theia-ide.org/). Could you please get the latest TheiaBlueprint version from https://theia-ide.org/docs/blueprint_download, start it and let us know if you can reproduce the defect? Does the <kbd>1</kbd> key work with your keyboard layout in the TheiaBlueprint app? Thank you!

samtd, 2022-12-13:
> 
@kittaakos, i tested with Theia IDE and works.
Theiablueprint cloud: Work
Theiablueprint installed v1.32.0: Work
arduino-ide_2.0.3_Windows_64bit.zip  : Dont Work
arduino-ide_2.0.3_Windows_64bit.exe :  Dont Work
Arduino Web App Version - 6.4.6: Work

Edição	Windows 11 Pro
Versão	22H2
Instalado em	‎08/‎10/‎2022
Compilação do SO	22621.900

If i use key number 1 with shift it works the correct character "!", but number 1 i can't have
all other keys also works

kittaakos, 2022-12-14:
Thanks for the investigation, @samtd. Please open a separate issue in this repo. Otherwise, the valuable information you collected might get lost in this closed issue. Thank you!

kittaakos, 2022-12-15:
> open a separate issue

https://github.com/arduino/arduino-ide/issues/1772

PR #1478 (2022-09-26): #989 Listen on keyboard layout changes from the OS
### Motivation

Delegate keyboard layout changes from the OS to the IDE2 frontend.

How to test:
 - Follow the steps from #989.

### Change description

### Other information

Closes #989

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / updated (for bug fixes / features)
commit 707f3bef (2022-09-26): Listen on keyboard layout changes from the OS.

Closes #989