arduino/arduino-ide #1001: Permission Request for distributing arduino ide v2 on flathub

joshuag1000, 2022-05-21:
### Describe the request

I wasn't sure where to ask this but I have been working on a flatpak version of the new ide so it can be distributed using flathub.
One of the requirements for creating the new app request is to get developer permission to submit the app to flathub.
So basically is this ok for me to do? I'd like to assume yes because v1 is already on flathub but I thought i should double check.
(i mentioned the flatpak build here: #356).

Details about the flatpak can be seen here: https://github.com/flathub/flathub/pull/3176

SuperNinja.

### Describe the current behavior

N/A

### Arduino IDE version

ALL

### Operating system

Linux

### Operating system version

ANY

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

ubidefeo, 2022-05-26:
hi @joshuag1000 

Thank you for this request, I'll defer to @alranel for a response

u.

alranel, 2022-05-26:
Hi @joshuag1000, you have our permission (and gratitude). Thank you!

closed 2022-05-26 by alranel

smith558, 2022-12-18:
The upload is not working for Arduino IDE v2.

Adding this to `50-arduino.rules` doesn't fix the issue either.
```
KERNEL=="ttyUSB[0-9]*",MODE="0666"
KERNEL=="ttyACM[0-9]*",MODE="0666"
```