arduino/arduino-ide #1005: Multible spaces converted to single space

Akinollo, 2022-05-24:
### Describe the problem

Like alredy closed issue #675:
Multible spaces are converted in one single space. Formated output via serial monitor impossible.

### To reproduce

  printf("><\n");
  printf("> <\n");
  printf(">  <\n");
  printf(">   <\n");
  printf(">    <\n");
  printf(">     <\n");
  printf(">      <\n");
produces:
22:55:50.387 -> ><
22:55:50.387 -> > <
22:55:50.387 -> > <
22:55:50.387 -> > <
22:55:50.387 -> > <
22:55:50.387 -> > <
22:55:50.387 -> > <

### Expected behavior

keep format of strings!!
22:55:50.387 -> ><
22:55:50.387 -> > <
22:55:50.387 -> >  <
22:55:50.387 -> >   <
22:55:50.387 -> >    <
22:55:50.387 -> >     <
22:55:50.387 -> >      <

### Arduino IDE version

arduino-ide_2.0.0-rc6-nightly-20220519

### Operating system

Windows

### Operating system version

10

### Additional context

Issue was solve an closed already.
Will this ever work?

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

Akinollo, 2022-05-24:
Funny, even in this window the multiple spaces disappear.... An editor that edits the text itself is useless...

per1234, 2022-05-24:
Hi @Akinollo. Thanks for your report.

I will need some more information before I can investigate this issue:

Please provide a complete minimal Arduino sketch I can use to reproduce the problem.

Please tell me which Arduino board you have selected in the Arduino IDE. If it is a 3rd party board, please provide instructions I can follow for installing support in the Arduino IDE (e.g., Boards Manager URL, etc.)

Akinollo, 2022-05-25:
Hi per1234,
Seems to be solved again with newest version of today;
arduino-ide_nightly-20220525_Windows_64bit
I used
arduino-ide_nightly-20220519_Windows_64bit
before

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  while(1)
  {
    Serial.println("><");
    Serial.println("> <");
    Serial.println(">  <");
    Serial.println(">   <");
    Serial.println(">    <");
  }
}

per1234, 2022-05-25:
Thanks for checking. I'm glad it is working now. Please let us know if you encounter any other problems.