arduino/arduino-ide #1006: Serial monitor goes out of focus

intoku, 2022-05-26:
### Describe the problem

Whenever compiling, the "Output" tab always comes to focus, even if the serial monitor is out; Trying to fully switch off the "Output" tab was not possible; 
The inability to keep the Serial to be sticky, forces the user to additionally click every time after compiling, and this can be nerve-wrecking while debugging.

Probably just an option to disable the "Output" tab, or remove the automatic focus - could solve this issue

### To reproduce

Switch on serial monitor. Compile sketch; Serial monitor tab is disabled, the "Output" tab is active.

### Expected behavior

Serial should be visible while debugging:

### Arduino IDE version

2.0.0.rc6

### Operating system

macOS

### Operating system version

catalina

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-05-27:
Thanks for your request @intoku. A proposal that would also resolve this via a different approach is tracked here: https://github.com/arduino/arduino-ide/issues/289

cshamis, 2022-10-28:
Yeah, those are cool, but a simple "pin" window to front would answer a lot of folks needs.

per1234, 2023-03-29:
Closing as fixed by https://github.com/arduino/arduino-ide/pull/1979

closed 2023-03-29 by per1234