arduino/arduino-ide #1009: Use code formatter tool directly

per1234, 2022-05-27:
### Describe the request

Use the code formatting tool directly for the IDE's "Auto Format" capability.

🙂 This will ensure the fastest possible response when the user triggers a code formatting operation.

### Describe the current behavior

The Arduino IDE provides a sketch code formatter (e.g., **Tools > Auto Format**).

Formatting is currently done via [the "**Arduino Language Server**"](https://github.com/arduino/arduino-language-server), which also provides the code aware features such as autocompletion, problem detection, "Go to definition". This approach is convenient because [a high quality code formatting capability](https://clangd.llvm.org/features#formatting) is already available is the ["**clangd**" C++ language server](https://clangd.llvm.org/) used by **Arduino Language Server**.

The language server features require the complete program to be known, which includes far more than the code in the sketch itself (i.e., toolchain, core, libraries). The amount of processing involved is significant, which means that there will always be some latency in the language server results.

Code formatting is different in that it only requires passing the text of the current sketch editor tab through the formatter tool, with no program processing needed.

🙁 The auto format process is unnecessarily subject to program processing latency, which causes significant delays between the user triggering an "Auto Format" operation and getting the result.

### Arduino IDE version

2.0.0-rc6-snapshot-522a5c6

### Operating system

Windows, Linux, macOS

### Operating system version

Any

### Additional context

Build infrastructure for [the standalone ClangFormat code formatter tool](https://clang.llvm.org/docs/ClangFormat.html) is already in place at https://github.com/arduino/clang-static-binaries

---

Please let's start with a PoC and see how tricky this is

---

- [ ] @per1234 to provide .clang-format configuration file

---

Unfortunately the main limitation is that clang-format will not accept stdin as input but requires a file, so when formatting a Tab's content we'll need to use a temp file and run it as follows

```text
clang-format --style=/path/to/style-config.clang-format /path/to/source-file.ino
```

---

This proposal is equivalent to the approach taken by Arduino IDE 1.x

---

Related:

- https://github.com/arduino/arduino-ide/issues/337
- https://github.com/arduino/arduino-ide/issues/396

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-05-27:
> @per1234 to provide .clang-format configuration file

The [**ClangFormat** configuration](https://clang.llvm.org/docs/ClangFormatStyleOptions.html#configuring-style-with-clang-format) is here in the `config` variable (you can just copy/paste the content of that string):

https://github.com/arduino/arduino-language-server/blob/main/ls/ls_formatter.go

kittaakos, 2022-06-02:
PR: https://github.com/arduino/arduino-ide/pull/1019. Please review. Thank you!

ubidefeo, 2022-06-06:
LGTM

commit fce1babc (2022-06-07): Use `clang-format` as the default sketch formatter.

 - Bumped `clangd` to `14.0.0`,
 - Can use `.clang-format` from:
   - current sketch folder,
   - `~/.arduinoIDE/.clang-format`,
   - `directories#data/.clang-format`, or
   - falls back to default formatter styles.

Closes #1009
Closes #566
PR #1019 (2022-06-07): #1009, #566: Use `clang-format` as the default sketch formatter.
### Motivation

Use `clang-format` as the default sketch file formatter. It's faster and does not require a running LS.

### Change description

 - Bumped `clangd` to `14.0.0`,

### Other information

Closes #1009
Closes #566

For the reviewers:
 - Please open dev tools for verification. _Formatting `$URI` [Range: `$RANGE`]_ should be logged when the `clang-format`-ter is used instead of the LS.
 - The formatter should work if the LS is not running.
[image]

(👆 The _Format Document_ and _Format Selection_ context menu items are a [... cut]
commit a59e0da2 (2022-06-07): Use `clang-format` as the default sketch formatter.

 - Bumped `clangd` to `14.0.0`,
 - Can use `.clang-format` from:
   - current sketch folder,
   - `~/.arduinoIDE/.clang-format`,
   - `directories#data/.clang-format`, or
   - falls back to default formatter styles.

Closes #1009
Closes #566
closed 2022-06-07 by kittaakos