arduino/arduino-ide #1010: In Windows, need support for longer package paths or ability to specify package folder

MikeFarrington, 2022-05-29:
### Describe the problem

My windows user name, "Michael Farrington", is 18 characters long.  Windows applications, by default, can only support 260 path characters.  Due to long package names, combined with Arduino placing files into %LocalAppData%, makes the maximum path to a library effectively much shorter.

Windows 10 and above now support breaking the 260 character path limit, but applications must be tagged in their application manifest with the "longPathAware" attribute.  See this [Microsoft Docs link](https://docs.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation) for more information.

I tried creating my own manifest files to the IDE executable, and executables in the arduino-ide-extensions\build folder, but that did not work.  I then tried to do the same to the cc1plus.exe ESP32 compiler executables.  Nothing worked, and am left unable to compile some projects (using 1.8.19 or 2.0 RC6), unless I create a new windows profile with a shorter username.

Below is an example of the error.  The file that it says is not found is indeed in the folder.  If I could specify the folder in which to store the packages, I could gain up to 57 extra characters of path length (if I go to the extreme of placing packages in folder "c:\p").  Obviously removing Windows path length limitations by using longPathAware is the better solution, but I imagine that would require

I realize that this is mostly an issue caused by Windows, but allowing users to move the packages folder location would be of great help.

```
In file included from c:\users\michael farrington\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\string:41,
                 from c:\users\michael farrington\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\michael farrington\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\array:39,
                 from c:\users\michael farrington\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\michael farrington\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\functional:54,
                 from C:\Users\Michael Farrington\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/HardwareSerial.h:49,
                 from C:\Users\Michael Farrington\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/Arduino.h:167,
                 from C:\Users\Michael Farrington\AppData\Local\Temp\arduino-sketch-ECB8BAD75F7407F9B3C84ADE862078A8\sketch\floppybird.ino.cpp:1:
c:\users\michael farrington\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory
 #include <bits/c++allocator.h> // Define the base class to std::allocator.
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### To reproduce

Try to compile a project that includes files with a file path longer than 260 characters.

### Expected behavior

All files to be found.

### Arduino IDE version

2.0.0-rc6-nightly-20220529

### Operating system

Windows

### Operating system version

Windows 11 (Version 21H1)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-05-29:
Thanks for taking the time to submit an issue @MikeFarrington .

I see we have another report about this issue as it affects the Arduino IDE at https://github.com/arduino/arduino-ide/issues/106.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, please feel free to comment in the other thread.

> I then tried to do the same to the cc1plus.exe ESP32 compiler executables. 

That is 3rd party software not maintained in this repository, so we couldn't help with that here.

>If I could specify the folder in which to store the packages

You can do that if you like. It is configured in the `directories.data` key of the Arduino CLI configuration file located in this file:

```text
c:\users\michael farrington\.arduinoIDE\arduino-cli.yaml
```

This configuration file is documented here:

https://arduino.github.io/arduino-cli/latest/configuration/

MikeFarrington, 2022-05-31:
Thanks, I was diving through all of the config files I could find in %localappdata%, but I missed that ".arduinoIDE" in the %userprofile% folder.  That did the trick, though it took two restarts of the IDE to take effect for some reason.  Perhaps those path settings should be surfaced via the preferences dialog.  I feel like a dummy for not reading 106 that you linked, I saw the topic name in google searches and figured it wasn't what I was looking for -- my apologies for wasting your time in that regard.  But also, you have my sincerest gratitude for such a prompt and thorough answer.

closed 2022-05-31 by MikeFarrington