arduino/arduino-ide #1017: My sketch diasappeared

glistok, 2022-06-01:
### Describe the problem

I am using the 2.0rc6 version of the IDE.  I had opened and closed several of the example sketchs and had even modified a couple.  They always opened just as I had last used them when starting the IDE program.  Now, I created a file from scratch, a small one, and had opened and closed the editor several times.  Once, when I opened the editor, there was a different sketch loaded and my new sketch was gone.  I have looked in the arduino directory and done a search on a couple of unique words the sketch contained.  Nothing was found.  The sketch was just gone

### To reproduce

I cannot reproduce the incident.  Neither can I find the sketch file.

### Expected behavior

I expected to see the sketch I created when I reopened the editor.

### Arduino IDE version

2.0rc6

### Operating system

Windows

### Operating system version

11

### Additional context

I may have missed a message that told me to save my sketch before closing the editor.  I have not seen one, however.  Also, I do have an extensive background in electronics and programming, I am, however, new to arduino IDE.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-01:
Hi @glistok. Thanks for your report.

>I may have missed a message that told me to save my sketch before closing the editor. I have not seen one, however.

There was a bug that caused there to be no message: https://github.com/arduino/arduino-ide/issues/595, https://github.com/arduino/arduino-ide/issues/862

I'm happy to tell you that the bug was fixed a couple of hours ago: https://github.com/arduino/arduino-ide/pull/893

If you would like to try out the latest version of the Arduino IDE with the fix, the tester builds are available for download here:

https://github.com/arduino/arduino-ide/actions/runs/2420412792#artifacts

The fix will also be in the nightly build tomorrow (today's nightly build is from before the fix was made), which is available from these links:

https://github.com/arduino/arduino-ide#nightly-builds

Please let us know what your results are if you give it a try.

As for your lost sketch, you can search the temporary directory on your computer. Unsaved sketches are staged  in folders that look something like this:

```text
C:\Users\<user name>\AppData\Local\Temp\.arduinoIDE-unsaved202251-6272-1v0qrmq.gp08\sketch_jun1a
```

**ⓘ** If looking for it with your file browser or command line, note that the `C:\Users\<user name>\AppData` folder is hidden by default. On Windows "**File Explorer**", you can make it visible by opening the "**View**" menu, then checking the box next to "**☐ Hidden items**".

glistok, 2022-06-01:
Hello,

Thanks for the great help.  I found a copy of the sketch in C:\Users\Acer\AppData\Local\Temp\arduino-language-server708689814\sketch.

I am installing the fixed code now.

George

https://github.com/arduino/arduino-ide/actions/runs/2420412792#artifacts<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Farduino%2Farduino-ide%2Factions%2Fruns%2F2420412792%23artifacts&data=05%7C01%7C%7Cb5bf2ec43ccc4c59a48508da43c20c78%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637896797956013915%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=cgfHVoT1AwEpP5b2TTwNNmAAcp8ljY6%2BIYwZD%2Fgka%2FA%3D&reserved=0>

https://github.com/arduino/arduino-ide#nightly-builds<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Farduino%2Farduino-ide%23nightly-builds&data=05%7C01%7C%7Cb5bf2ec43ccc4c59a48508da43c20c78%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637896797956013915%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=ss5s35AN9xabIt7vBeMMkgG1b%2BHk15zc%2BI8zRi5VPGo%3D&reserved=0>