arduino/arduino-ide #1020: Arduino IDE 2.0 RC (2.0.0-RC6) starts slowly!

gittanker, 2022-06-02:
### Describe the request

I come from China. Starting Arduino IDE 2.0 RC is slow! Up to 5 minutes!
If you are disconnected from the network. It'll be activated in three seconds.
Each startup requires disconnecting from the network. This is very troublesome for me.
Hopefully the next version will solve this problem that has been bothering me.
I learned through the Internet that many people have the same problem as me: the startup is very slow.

### Describe the current behavior

China IP, system, Asia China
Version is Arduino IDE 2.0 RC (2.0.0-RC6)

### Arduino IDE version

 Arduino IDE 2.0 RC (2.0.0-rc6)

### Operating system

Windows

### Operating system version

Windows 11

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

ckuethe, 2022-06-09:
try the recently-released RC7. It's much better for me.

ubidefeo, 2022-06-13:
RC7 should be much faster.
Closing issue.
Notify us if you still have problems.

closed 2022-06-13 by ubidefeo

Sezh73ok, 2022-09-26:
Поставил версию Ардуино Иде 2.0.0 не запускается не на Виндоус 7,10,11
не знаю ,что делать . Туркменистан

per1234, 2022-09-26:
Hi @Sezh73ok. Please request assistance over on the Arduino forum. If you do not wish to use English, there may be an international category available for your preferred language:

https://forum.arduino.cc/c/international/10