arduino/arduino-ide #1023: Build process behaves strange

logowe, 2022-06-06:
### Describe the problem

Just to put it strait, the IDE builds and uploads executable code. 
Hence this report shouldn't be placed here?

I experience strange behavour when building:
1. A new session is started and the build is initiated: Works as expected.

2. The build is initiated a second time in this session. Some or no changes are done in the source code: 
a) Seems like the whole process now is done twice:
b) In the middle of the process (after finishing the first time?) the message pops up "Upload done" and the target system executes the program.
c) Then it still takes about the same time to finish the build process. At the end the upload is displayed by showing the % progress. The target system does not start a second time.
d) The program version the target system executes at 2.b) instant complies to the source code. 

3. If I close the IDE after 2. b) (while it still pretends to build) and start a new session, then start the build process it executes like 1.. All object files are in the cache from the previous session. After the linking, the upload is displayed by showing the % progress and the target system starts to execute the program.

4. Repeatetly building in the same session takes progressively longer time to finish. 
I experience up to factor 10 after about 20 builds.
This makes me close the IDE and start a new session to get back to normal build performance.

### To reproduce

1. Load some project consistiong of at least 5000 C++ code lines.
2. Build it repeatetly and observe whats happening.

### Expected behavior

The build process allways works the same way with the same performance. 
At the end of the build process the target system is flashed and starts to execute the program.

### Arduino IDE version

2.0.0 r6

### Operating system

Windows

### Operating system version

10

### Additional context

Espressif Board Manager ESP32 V.2.0.3.
DFROBOT Fire Beetle Board-ESP32 (V4.0)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-07:
Hi @logowe. Thanks for your report.

Please select **File > Preferences...** from the Arduino IDE menus and then tell me what are the states of the checkboxes next to the following settings:

- **Show verbose output during > ☐ compile**
- **Show verbose output during > ☐ upload**

logowe, 2022-06-07:
Ohh ... is this relevant?
See the screen shot, please
[image]
.

closed 2022-06-07 by logowe

per1234, 2022-06-07:
Thanks. Yes, it is relevant. There is a known bug with the throughput of the output panel.

- https://github.com/arduino/arduino-ide/issues/68
- https://github.com/arduino/arduino-ide/issues/405

This gives a false impression that the compilation and upload processes are going very slowly. The truth is that those processes have completed long before the output finishes trickling through the horrible bottleneck somewhere in the system.

---

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, but if you end up with additional information to share, feel free to comment in the other thread.

logowe, 2022-06-07:
Yes I agree to close this issue. 
I saw the other 2 issues but did not recognise them to be linked.  
In the meantime I will work with these 2 options disabled.
Thank you.