arduino/arduino-ide #1028: Theme

reigate, 2022-06-09:
### Describe the problem

When Preferences>Theme>Dark(Arduino) is selected, highlight colour is bright yellow making any selected text almost impossible to read.

### To reproduce

As above.

### Expected behavior

Change the highlight colour.

### Arduino IDE version

Version: 2.0.0-rc7 Date: 2022-06-09

### Operating system

Windows

### Operating system version

Windows 11 Pro

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-06-13:
[image]

For the record, this is with the new _Dark (Arduino)_ theme, not the high contrast one.

91volt, 2022-06-28:
Updated in the json tokens

[image]

[image]