arduino/arduino-ide #1030: changes to boards.txt are not picked up

KurtE, 2022-06-09:
### Describe the problem

I install in this case support for Teensy boards. 

I verify that it works and notice some of the options were not enabled in boards.txt.

So I go and edit the boards.txt to enable it.  So I then restart the IDE and my changes are not reflected in menu.

### To reproduce

Install support for Teensy: 
:update preferences with: https://www.pjrc.com/teensy/td_156/package_teensy_index.json

Install the boards, and then for example Choose Teensy 3.2 / 3.1

Look at the CPU speed menu item and see that 144mhz is not in the list.

Go edit the boards.txt file:
```
...
#uncomment these if you want to try faster overclocking
#teensy31.menu.speed.144=144 MHz (overclock)
#teensy31.menu.speed.168=168 MHz (overclock)
...
```
And lets say uncomment the 144 Mhz line. 
And save the changes

restart the IDE and then look at the CPU speeds for and the new item is not in the list. 

### Expected behavior

I would expect in this case the 144 mhz cpu speed to show up, like it does in Arduino 1.x

I am assuming you are caching the data out of the different tools.txt files and as such not being seen.  Maybe the cache code need to remember modification dates or the like.  

Alternatively need some mechanism to tell the IDE to refresh it's cache. 

### Arduino IDE version

RC7 which was released today

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

Note: I first noticed this when I was trying to update @PaulStoffregen package for TD1.56 to use the stuff from 1.57 Beta 2, so I was updating the  boards.txt to have an additional USB type for T4.x and soon T3.x

But used the CPU speed example as it is easier to reproduce. 

I have not yet checked to see if you pickup changes or the creation of the local.boards.txt file or not

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-10:
Thanks for the report @KurtE. This caching issue also affects the **Tools > Programmer** menu contents: https://github.com/arduino/arduino-ide/issues/591

I'll share the workaround:

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Delete the "**User data**" folder:
   - **Windows:**
     - ```text
       C:\Users\<user name>\AppData\Roaming\arduino-ide\
       ```
   - **Linux:**
     - ```text
       ~/.config/arduino-ide/
       ```
   - **macOS:**
     - ```text
       ~/Library/Application Support/arduino-ide/
       ```
1. Start the Arduino IDE.

The custom board options menus should now reflect any changes that were made to `boards.txt`.

KurtE, 2022-06-10:
@per1234 Thanks, that helps. 

I remember from before there was some location to delete.  

This time, so far it does not look like deleting this data caused the IDE to download all of the standard libraries again and wipe out one that was already there.  (https://github.com/arduino/arduino-ide/issues/798)

Thanks again

MrSniffer, 2022-10-10:
This issue is still not fixed in Arduino IDE Windows 2.0.0 x64.

joostbijl, 2022-11-02:
I can confirm the issue is still present in 2.0.1

sevelm, 2023-01-19:
i can not find this folder in Win10 x64
`C:\Users\<user name>\AppData\Roaming\arduino-ide\`
hidden files and folders are shown. i use IDE1.8.19
any ideas?

per1234, 2023-01-20:
>i use IDE1.8.19

That is the reason. This is a discussion about Arduino IDE 2.x. It is not applicable to the Arduino IDE 1.8.19 version you are using.

Please post on the Arduino forum to ask for assistance with your project:

https://forum.arduino.cc/c/using-arduino/installation-troubleshooting/18

I'm sure we'll be able to help you out over there.

Jackfritt, 2023-02-19:
Still present in 2.0.3

puzirev, 2023-04-08:
Same issue when adding new board type in stm32duino on 2.0.4. Deleting "User data" folder helps.

KurtE, 2023-05-10:
Still appears to be issue with 2.1 (actually nightly build 0510)

Wondering if deleting all of C:\Users\<user name>\AppData\Roaming\arduino-ide is still the best/only option to resolve this?

You still end up with lots of strange messages after the change, like:
```
Platform arduino:avr@1.8.6 already installed
Failed to install library: Arduino_BuiltIn:1.0.0.
Error: 9 FAILED_PRECONDITION: Library 'Arduino_BuiltIn@1.0.0' not found

```

efirmata, 2023-07-04:
Still have to delete the files...

KurtE, 2023-07-31:
Side Note: at least with the newer builds, changes to platform.txt are picked up, if you close out the IDE and restart it.

As mentioned in the form thread:
https://forum.arduino.cc/t/why-does-the-compiler-add-4x-00-to-a-string-array-element/1152890/57