arduino/arduino-ide #1031: GUI-based settings commands produce an empty interface

per1234, 2022-06-09:
### Describe the problem

In addition to the common settings that are configurable via the Arduino IDE's **File > Preferences** dialog, there are some additional settings that can be useful to advanced users, beta testers, or for troubleshooting problems with the Arduino IDE.

If the user happens to know the right setting key name, these can be adjusted by manually editing a `settings.json` file. Previously, it was also possible to do via a graphical interface, accessed via a command palette command.

🐛 The advanced settings graphical interface is now empty.

### To reproduce

1. Open the "**Command Palette**".
   (<kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> or <kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> by default)
1. Run the "**Preferences: Open Settings (UI)**" command.

🐛 There are no settings, so the interface is pointless.

[image]

### Expected behavior

Remove pointless and confusing command palette items.

**- OR -**

Restore the graphical advanced settings interface.

### Arduino IDE version

2.0.0-rc7

### Operating system

Windows

### Operating system version

10

### Additional context

Since the advanced settings will only be of interest to advanced users or users reporting problems with the IDE, who can be guided in their use by a knowledgeable helper, requiring manual editing of a JSON file to set them is perfectly reasonable, and in fact matches the approach used for the Arduino IDE 1.x advanced settings, which could only be configured by manually editing the `preferences.txt` file.

However, if this is to be done, the location and editing procedure must be documented. All configuration keys and their purpose must be clearly documented, as was already provided by the GUI-based settings interface.

---

I bisected the change to https://github.com/arduino/arduino-ide/commit/4c55807392b3b724e2c5912280c621f9ed8db5e6

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

serpinio, 2022-06-10:
Wait...

What was so horribly wrong with Open Settings (**UI**)? It was a very convenient place to fine-tune editor behavior, and who suddenly decided that "advanced settings will only be of interest to advanced users or users reporting problems with the IDE". Personally, while browsing the GUI I discovered and ended up changing _38_ various settings thanks to descriptions available right there, next to the setting name, which in turn improved my productivity with the editor.

Why remove this option altogether? Am I missing some big idea?
Please, restore Open Settings (**UI**), no really, _why remove something that worked just fine in 2.0.0-RC6_?

serpinio, 2022-06-12:
**Thank you!!!** You guys are awesome!

PR #1046 (2022-06-13): #1031 Restored the Settings UI. Deferred model loading.

### Motivation
This PR will restore the (advanced) Settings UI in the IDE2. The proposed changes also defer the model (and UI) loading until the IDE2 is ready. Delaying the model load keeps the startup's performance OK and the Settings UI feature in IDE2.

### Change description

How to verify:
 - Open the IDE2, and open the preferences as described in #1031,
 - Verify that the preferences UI is the same as before `rc7`,
 - Close the IDE2, and reopen it.
 - The widget is open, the content is unavailable, then the UI refreshes. Verify tha [... cut]
commit a8047660 (2022-06-13): Restored the Settings UI. Deferred model loading.

Closes #1031
closed 2022-06-13 by kittaakos