arduino/arduino-ide #1036: No boards available

ChirpyMisha, 2022-06-10:
### Describe the problem

After updating to the latest release (Arduino IDE 2.0.0-rc7) there is no board available in the boards manager or when selecting a port. Neither the default Arduino boards nor the boards from the Additional boards manager URLs are available. 

This image shows that both lists are empty. 
[image]

### To reproduce

- Open the Arduino IDE
- Go to the boards manager

Or:
- Open the Arduino IDE
- Select the dropdown menu with "X No board selected"
- Select the correct port from this list

### Expected behavior

A list of available boards should appear in the designated fields.

### Arduino IDE version

Version: 2.0.0-rc7-nightly-20220610

### Operating system

Windows

### Operating system version

21H1 

### Additional context

Restarting the IDE did not solve the issue. Installing the nightly build also didn't solve the issue.

Additional boards manager URLs: http://arduino.esp8266.com/stable/package_esp8266com_index.json,https://dl.espressif.com/dl/package_esp32_index.json
(removing these and restarting the IDE also didn't solve the issue)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-10:
Hi @ChirpyMisha. Thanks so much for your report.

I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Start Windows "**File Explorer**".
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type the following command:
   ```text
   & ".\Arduino IDE"
   ```
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as the Arduino IDE starts (or fails to start).
1. Wait for the Arduino IDE to finish starting, or failing to start.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

ChirpyMisha, 2022-06-10:
PS C:\Program Files\Arduino IDE> .\Arduino IDE
.\Arduino : The term '.\Arduino' is not recognized as the name of a cmdlet, function, script file, or operable program.
 Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ .\Arduino IDE
+ ~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (.\Arduino:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

PS C:\Program Files\Arduino IDE> & ".\Arduino IDE"
PS C:\Program Files\Arduino IDE>
Starting backend process. PID: 192464
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/misha/.theia'
root INFO Backend Object.initialize: 5.3 ms [Finished 1.543 s after backend start]
root INFO Backend Object.configure: 4.0 ms [Finished 1.543 s after backend start]
root INFO Backend MessagingContribution.onStart: 1.8 ms [Finished 1.543 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.5 ms [Finished 1.601 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.4 ms [Finished 1.601 s after backend start]
root WARN Backend LocalizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 58.1 ms [Finished 1.602 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.6 ms [Finished 1.602 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.4 ms [Finished 1.602 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 2.6 ms [Finished 1.604 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 2.5 ms [Finished 1.604 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 2.9 ms [Finished 1.605 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.6 ms [Finished 1.607 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 0.5 ms [Finished 1.607 s after backend start]
daemon INFO Starting daemon from C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
root INFO Backend NodeFileUploadService.configure: 3.0 ms [Finished 1.609 s after backend start]
root INFO Backend PlotterBackendContribution.initialize: 0.4 ms [Finished 1.610 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.4 ms [Finished 1.610 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 9.6 ms [Finished 1.621 s after backend start]
root INFO Backend PluginApiContribution.configure: 1.3 ms [Finished 1.623 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.5 ms [Finished 1.625 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.2 ms [Finished 1.625 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 0.1 ms [Finished 1.626 s after backend start]
root INFO Theia app listening on http://localhost:61170.
root INFO Finished starting backend application: 9.4 ms [Finished 1.630 s after backend start]
root WARN The local plugin referenced by local-dir:/c%3A/Users/misha/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/misha/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\misha\.arduinoProIDE\plugins does not exist.
Restoring workspace roots: c:\Users\misha\OneDrive\Persoonlijk\Programming\Arduino\ESP-01 projects\MCP9808-Wifi-Temp-Sensor
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-nl" to a VS Code extension "vscode-language-pack-nl@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-tr" to a VS Code extension "vscode-language-pack-tr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... cut ...]
root INFO [d8142ade-925f-45da-a2c5-5768219291d9][ms-ceintl.vscode-language-pack-ja]: Started plugin.
root INFO [d8142ade-925f-45da-a2c5-5768219291d9][ms-ceintl.vscode-language-pack-nl]: Started plugin.
root INFO [d8142ade-925f-45da-a2c5-5768219291d9][ms-ceintl.vscode-language-pack-tr]: Started plugin.
root INFO [d8142ade-925f-45da-a2c5-5768219291d9][ms-ceintl.vscode-language-pack-zh-hans]: Started plugin.
root INFO [d8142ade-925f-45da-a2c5-5768219291d9] Start of 11 plugins: 1286.1 ms [Finished 6.664 s after frontend start]

per1234, 2022-06-10:
## Problem

The problem is indicated by this line of the output:

```text
root ERROR Error loading hardware platform: loading platform release esp8266:esp8266@2.5.2: loading boards: skipping loading of boards esp8266:esp8266:espduino: malformed custom board options
```

Older versions of the ESP8266 Arduino boards platform had a configuration for the "ESPDuino" board which is not compatible with Arduino IDE 2.x. Recent versions of the IDE were resilient to this problem, but unfortunately that resiliency was lost in the 2.0.0-rc7 release (by https://github.com/arduino/arduino-ide/pull/1018 to be more precise).

Fortunately, the configuration of the ESP8266 boards platform was updated last year to a configuration which is fully supported by Arduino IDE 2.x (https://github.com/esp8266/Arduino/pull/8153). However, you are using an old version that still has the unsupported configuration.

## Workaround

I think the easiest workaround for you to get back up and running with 2.0.0-rc7 will be to temporarily use an older version of the Arduino IDE to update your ESP8266 boards platform. I'll provide instructions:

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Click this link to download version 2.0.0-rc6 of the Arduino IDE:
   https://github.com/arduino/arduino-ide/releases/download/2.0.0-rc6/arduino-ide_2.0.0-rc6_Windows_64bit.zip
1. Wait for the download to finish.
1. Unzip the downloaded file.
1. Run the file `Arduino IDE.exe` from the unzipped folder.
1. Select **Tools > Board > Boards Manager**" to open the "**Boards Manager**" view in the side panel if it is not already open.
1. Find "**esp8266 by ESP8266 Community**" in the list of boards platforms. Click on it.
1. Make sure "**3.0.2**" is selected from the version menu on the esp8266 platform listing.
1. Click the <kbd>**INSTALL**</kbd> button.
1. Wait for the install operation to finish.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start Arduino IDE 2.0.0-rc7.

Hopefully 2.0.0-rc7 will now be working. As a bonus, you will have the latest and greatest version of the ESP8266 boards platform installed.

You are now welcome to delete the 2.0.0-rc6 ZIP file and the unzipped folder. It was only needed for the sake of updating the ESP8266 boards platform.

## Arduino IDE defect

I bisected the introduction to 4c55807 (it does not occur at eaf14aa).

Since 4c55807 bumped the Arduino IDE's Arduino CLI dependency from 0.21.0 to 0.23.0, my hunch was that the defect was actually a regression in Arduino CLI, which had problems with handling these configurations (introduced by https://github.com/arduino/arduino-cli/pull/1274 I believe) before it was made more resilient via https://github.com/arduino/arduino-cli/pull/1326.

However, that is only a guess. It might have also been caused by one of the extensive changes made to the Arduino IDE code base via 4c55807.

kittaakos, 2022-06-10:
If somebody helps me reproduce the issue with RC-7, I am more than happy to narrow down the scope of the problem.

per1234, 2022-06-10:
Here are the instructions @kittaakos:

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Add the following URL into the "**Additional Boards Manager URLs**" field.
   ```text
   https://arduino.esp8266.com/stable/package_esp8266com_index.json
   ```
1. Click the **<kbd>OK</kbd>** button.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the Arduino IDE.
   **ⓘ** This restart is required due to https://github.com/arduino/arduino-ide/issues/637
1. Open the "**Boards Manager**" view.
1. Find "**esp8266 by ESP8266 Community**" in the list of boards platforms. Click on it.
1. Select "**3.0.0**" from the version menu on the esp8266 platform listing.
   **ⓘ** **3.0.1** is the minimum supported version of the platform, so it is not necessary to go all the way back to the super old **2.5.2** version **ChirpyMisha** was using.
1. Click the <kbd>**INSTALL**</kbd> button.
1. Wait for the install operation to finish.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the Arduino IDE.

🐛 The IDE exhibits the symptoms reported by **ChirpyMisha**.

ChirpyMisha, 2022-06-10:
This has solved the issue for me. Thank you very much!

KurtE, 2022-06-10:
Quick note: I ran into this problem a couple times in the last few days, on two different machines.  One my main dev machine, and it probably was the esp8266 stuff mentioned.

But I also run into it this morning on Fresh install, (RPI4 Ubuntu 22.04) using install from: https://github.com/koendv/arduino-ide-raspberrypi

I installed it and ran it first time.  and it showed up the boards manager and did the downloads of libraries. 

I Then tried to install the Teensy install by adding it to the additional boards Manager URLS

I meant to type: https://www.pjrc.com/teensy/td_156/package_teensy_index.json

But typed instead: https://www.pjrc.com/td_156/package_teensy_index.json

I exited IDE restarted it and the boards list was empty.

I fixed the additional URLS to be correct, restarted the IDE and the boards list worked.

per1234, 2022-06-10:
>I meant to type: https://www.pjrc.com/teensy/td_156/package_teensy_index.json
>
>But typed instead: https://www.pjrc.com/td_156/package_teensy_index.json

Thanks for the report @KurtE. I can reproduce that (any URL that 404s e.g., `https://example.com/package_foo_index.json` will do the trick) and also can bisect it to https://github.com/arduino/arduino-ide/commit/4c55807392b3b724e2c5912280c621f9ed8db5e6.

I do suspect it has the same cause as the problem reported by **ChirpyMisha** (failure on recoverable error while initializing Arduino CLI via the gRPC interface).

commit 4f0f3e81 (2022-06-11): Relaxed the error handling of the core client init

For example, `malformed custom board options` was incorrectly detected
as loading JSON index file error.

Closes #1036
commit 7c2843f7 (2022-06-13): Relaxed the error handling of the core client init

For example, `malformed custom board options` was incorrectly detected
as loading JSON index file error.

Closes #1036