arduino/arduino-ide #1038: [sketchbook] The view shows broken hierarchy when opening the MarlinFirmware/Marlin project

kittaakos, 2022-06-10:
### Describe the problem

The view shows some strange sketchbook hierarchy when I have [Marlin](https://github.com/MarlinFirmware/Marlin) project opened in IDE2.

[image]

The actual FS content:
```
% ls -al
total 0
drwxr-xr-x  14 a.kitta  staff  448 Jun 10 16:43 .
drwx------+  5 a.kitta  staff  160 May 27 15:32 ..
drwxr-xr-x  21 a.kitta  staff  672 Jun 10 16:43 Marlin
drwxr-xr-x@  4 a.kitta  staff  128 Jun 10 16:32 MultipleFiles
drwxr-xr-x@  3 a.kitta  staff   96 Jun  9 14:09 libraries
drwxr-xr-x   4 a.kitta  staff  128 Jun  2 14:31 sketch with space
drwxr-xr-x   3 a.kitta  staff   96 Jun 10 13:11 sketch_jun10a
drwxr-xr-x   3 a.kitta  staff   96 Jun  1 14:56 sketch_jun1a
drwxr-xr-x   6 a.kitta  staff  192 Jun 10 09:37 sketch_jun1b
drwxr-xr-x   3 a.kitta  staff   96 Jun  3 15:11 sketch_jun3a
drwxr-xr-x   4 a.kitta  staff  128 Jun 10 14:12 sketch_jun3b
drwxr-xr-x   3 a.kitta  staff   96 May 27 14:31 sketch_may27a
drwxr-xr-x   3 a.kitta  staff   96 May 27 14:31 sketch_may27b
drwxr-xr-x@  3 a.kitta  staff   96 May 27 14:31 sketch_may27c

```

### To reproduce

 - Clone https://github.com/MarlinFirmware/Marlin to the `directories#user` location (on macOS this is `~/Documents/Arduino`),
 - Open Marlin in IDE2

### Expected behavior

I am not sure what to expect but the sketchbook tree looks incorrect. If there is a complex sketch, IDE2 should show the sketch `Marlin` as any other sketches.

### Arduino IDE version

Version: 2.0.0-rc7 Date: 2022-06-09T08:19:19.877Z CLI Version: 0.23.0 [899dc91b]  Copyright © 2022 Arduino SA

### Operating system

macOS

### Operating system version

12.3.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-10:
>IDE2 should show the sketch Marlin as any other sketches.

It does. Right here:

[image]

The folder you see here is a parent folder which happens to have the same name as the sketch folder (the sketch is in [the `Marlin` subfolder of the repository](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x/Marlin)):

[image]

So the sketchbook tree looks like this:

```text
Arduino/
├── Marlin/
│   ├── LICENSE
│   ├── Makefile
│   ├── Marlin/
│   │   ├── Configuration.h
│   │   ├── Configuration_adv.h
│   │   ├── Makefile
│   │   ├── Marlin.ino
│   │   ├── Version.h
│   │   ├── lib/
│   │   └── src/
│   ├── README.md
│   ├── buildroot/
│   ├── config/
│   ├── docker/
│   ├── docker-compose.yml
│   ├── docs/
│   ├── get_test_targets.py
│   ├── ini/
│   ├── platformio.ini
│   └── process-palette.json
├── MultipleFiles/
└── [...]
```

(note the location of the `.ino` file at `Arduino/Marlin/Marlin/Marlin.ino`)

kittaakos, 2022-06-11:
Thanks! I close this issue although the indentation or UX could be improved.

Current:

[image]

Paintbrush edited version:

[image]

closed 2022-06-11 by kittaakos

per1234, 2022-06-11:
>the indentation or UX could be improved

Definitely! I also find it to be interpret the tree structure of a nested sketchbook. There is already an item in the private backlog about that (5303). I'll publish that issue to GitHub ASAP so we can track it more effectively.