arduino/arduino-ide #1042: Certain errors in libraries can stop the language server

PaulStoffregen, 2022-06-10:
### Describe the problem

I encountered an error which causes the language server to stop running.

### To reproduce

Using Arduino IDE 2.0.0-rc7.

1: add https://www.pjrc.com/teensy/td_156/package_teensy_index.json to additional board manager URLs in File > Preferences

2: in boards manager, search for "teensy" and install Teensy package 1.56.1

3: set the current board to Tools > Board > Teensy > Teensy 4.1

4: copy this code into the IDE editor

``` c
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.printf("test\n");
  delay(1000);
}
```

5: hover your mouse over "printf" and this popup with info from the language server should appear.
[image]

6: quit the IDE, and edit ~/.arduino15/packages/teensy/hardware/avr/1.56.1/cores/teensy4/Print.h

7: on line 115, delete the semicolon at the end of the line and save the file with this intentional error
[image]
[image]

8: restart the IDE and again hover your mouse over "printf", same as in step 5 - but not the popup does not appear.  Hovering over other keywords also gives no results, and other language server feature no longer work.
[image]

### Expected behavior

Even though the Print.h code has an error, one would expect the language server to give some guidance about the error, at at least continue running.  It shouldn't just stop and no longer function, just because code as a missing semicolon.

### Arduino IDE version

2.0.0-rc7

### Operating system

Linux

### Operating system version

Ubuntu 20.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

marcusobrien, 2023-06-06:
I am seeing this also on my sketch when trying to use the lvgl GUI lib and ESP32S3 etc. Dont know what is causing this, but it would be helpful if for example the error told you the file that was cuasing it so we casn actually investigate it

"Error communicating with the language server: Error: write EOF: undefined."

Can you now make the error "Error communicating with the language server: Error: write EOF: undefined in file %s"

TzOk83, 2023-06-09:
I'm experiencing the same issue:
"Error communicating with the language server: Error: write EOF: undefined."

The splash screen is also missing.

per1234, 2023-06-11:
Hi @marcusobrien and @TzOk83. Thanks for your reports. I'm going to ask you to provide the debug logs that are produced by the "**Arduino Language Server**" tool.

You can do that by following these instructions:

## A. Enable Language Server Logging

1. Use the <kbd>Ctrl</kbd>/<kbd>⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd> keyboard shortcut to open the "**Command Palette**" in Arduino IDE.
1. Select the "**Preferences: Open Settings (UI)**" command from the "**Command Palette**".
   The "**Preferences**" tab will now open in the editor panel.
1. In the "**Search Settings**" field, type `arduino.language.log`
1. Under the "**Arduino > Language: Log**", section, check the box next to "**☐ True if the Arduino Language Server should generate log files into the sketch folder. Otherwise, false. It's false by default.**"
1. Close the **Preferences** tab by clicking its **X** icon.

## B. Gather Logs

1. Perform whatever procedure causes this "`Error: write EOF: undefined` error in Arduino IDE.
1. Select **Sketch > Show Sketch Folder** from the Arduino IDE menus.
   The folder containing the sketch will open in your file manager.
1. ZIP the entire sketch folder.
   - **Instructions for Windows**:
     1. Right click the sketch folder.
     1. From the context menu that opens, select "**Send to**".
     1. Select "**Compressed (zipped) folder**" from the "**Send to**" menu.
   - **Instructions for Linux**:
     1. Right click the sketch folder.
     1. From the context menu that opens, select "**Compress...**".
     1. In the "**Create Archive**" dialog that opens, click the <kbd>Create</kbd> button.
   - **Instructions for macOS**:
     1. Control-click the sketch folder.
     1. From the shortcut menu that opens, select "**Compress**"
1. Attach the "ZIP" file that was created to a comment here on this GitHub issue.
   ⚠ Don't use an email reply to make the comment. You must make the comment via the web interface here on the GitHub website in order for the attachment to be made:
   [`https://github.com/arduino/arduino-ide/issues/1042`](https://github.com/arduino/arduino-ide/issues/1042)
   You can do this by drag and dropping the file onto the comment field, or by clicking the bar at the bottom of the comment interface.

## C. Disable Language Server Logging

When language server logging is enabled, log files are saved to every sketch you open in Arduino IDE and they steadily grow in size whenever you are editing the sketch. So the logging is something you would usually want to disable whenever generation of logs is not needed.

1. Use the <kbd>Ctrl</kbd>/<kbd>⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd> keyboard shortcut to open the "**Command Palette**" in Arduino IDE.
1. Select the "**Preferences: Open Settings (UI)**" command from the "**Command Palette**".
   The "**Preferences**" tab will now open in the editor panel.
1. In the "**Search Settings**" field, type `arduino.language.log`
1. Under the "**Arduino > Language: Log**", section, uncheck the box next to "**`☑` True if the Arduino Language Server should generate log files into the sketch folder. Otherwise, false. It's false by default.**"
1. Close the **Preferences** tab by clicking its **X** icon.

You are now welcome to delete the `.log` files that were added to the sketch folder.

TzOk83, 2023-06-11:
The sketch is irrelevant, it might even be an empty sketch.

the error is:
```
Started logging.
13:26:08.548476 Language server launched with arguments:
13:26:08.548476   arg[0] = d:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-language-server.exe
13:26:08.548476   arg[1] = -clangd
13:26:08.548476   arg[2] = d:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\clangd.exe
13:26:08.548476   arg[3] = -cli-daemon-addr
13:26:08.548476   arg[4] = localhost:1817
13:26:08.548476   arg[5] = -cli-daemon-instance
13:26:08.548476   arg[6] = 1
13:26:08.548476   arg[7] = -fqbn
13:26:08.548476   arg[8] = Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd
13:26:08.548476   arg[9] = -skip-libraries-discovery-on-rebuild
13:26:08.548476   arg[10] = -board-name
13:26:08.548476   arg[11] = "Generic STM32F103C series"
13:26:08.549019   arg[12] = -no-real-time-diagnostics
13:26:08.549019   arg[13] = -log
13:26:08.549019   arg[14] = -logpath
13:26:08.549019   arg[15] = d:\Tomek\Documents\Arduino\LangSrvTest
13:26:08.549019 logging to d:\Tomek\Documents\Arduino\LangSrvTest\inols.log
13:26:08.553397 [97mLS: : Initial board configuration: Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd[0m
13:26:08.553397 [97mLS: : arduino-language-server.exe Version: 0.7.4 Commit: d620b0b Date: 2023-01-11T15:42:52Z[0m
13:26:08.553397 [97mLS: : Language server build path: C:\Users\Tomek\AppData\Local\Temp\arduino-language-server1605439801[0m
13:26:08.553397 [97mLS: : Language server build sketch root: C:\Users\Tomek\AppData\Local\Temp\arduino-language-server1605439801\sketch[0m
13:26:08.553397 [97mLS: : Language server FULL build path: C:\Users\Tomek\AppData\Local\Temp\arduino-language-server3532616197[0m
13:26:08.553397 IN Elapsed: 0s
13:26:08.554324 [92mIDE --> LS REQU initialize 0[0m
13:26:08.580287 [92m                initialize 0: [93mwrite-locked[0m[0m
13:26:08.580287 [92m                initialize 0: [93mwrite-unlocked[0m[0m
13:26:08.581301 [96mINIT --- : initializing workbench: file:///d%3A/Tomek/Documents/Arduino/LangSrvTest[0m
13:26:08.581301 [96mINIT --- : [93mread-locked[0m[0m
13:26:08.581301 [96mINIT --- : [93mread-unlocked[0m[0m
13:26:08.581301 [92m                initialize 0: initialization parameters: {"capabilities":{"textDocumentSync":{"openClose":true,"change":2,"save":{"includeText":true}},"completionProvider":{"triggerCharacters":[".","\u003c","\u003e",":","\"","/"],"allCommitCharacters":[" ","\t","(",")","[","]","{","}","\u003c","\u003e",":",";",",","+","-","/","*","%","^","\u0026","#","?",".","=","\"","'","|"],"completionItem":{}},"hoverProvider":{},"signatureHelpProvider":{"triggerCharacters":["(",","]},"definitionProvider":{},"documentHighlightProvider":{},"documentSymbolProvider":{},"codeActionProvider":{"codeActionKinds":["quickfix","refactor","info"]},"documentFormattingProvider":{},"documentRangeFormattingProvider":{},"documentOnTypeFormattingProvider":{"firstTriggerCharacter":"\n"},"renameProvider":{},"executeCommandProvider":{"commands":["clangd.applyFix","clangd.applyTweak"]},"workspaceSymbolProvider":{}},"serverInfo":{"name":"arduino-language-server","version":"0.7.4"}}[0m
13:26:08.581301 [32mIDE <-- LS RESP initialize 0[0m
13:26:08.581301 OUT Elapsed: 0s
13:26:08.583316 IN Elapsed: 2.0155ms
13:26:08.583316 [92mIDE --> LS NOTIF initialized[0m
13:26:08.583316 [92m                 initialized: Notification is not propagated to clangd[0m
13:26:08.583316 IN Elapsed: 0s
13:26:08.583316 [92mIDE --> LS NOTIF textDocument/didOpen[0m
13:26:08.583316 [92m                 textDocument/didOpen: [93mwrite-locked[0m[0m
13:26:08.583316 [92m                 textDocument/didOpen: (throttled: waiting for clangd)[0m
13:26:08.583316 [92m                 textDocument/didOpen: [93munlocked (waiting clangd)[0m[0m
13:26:08.894911 [96mINIT --- : Running build with: {
  "instance": {
    "id": 1
  },
  "fqbn": "Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd",
  "sketch_path": "D:\\Tomek\\Documents\\Arduino\\LangSrvTest",
  "build_path": "C:\\Users\\Tomek\\AppData\\Local\\Temp\\arduino-language-server3532616197",
  "verbose": true,
  "create_compilation_database_only": true
}[0m
13:26:08.928111 [96mINIT --- : build stdout:[0m
13:26:08.928308 [96mINIT --- : [0m
13:26:08.928308 [96mINIT --- : build stderr:[0m
13:26:08.928308 [96mINIT --- : [0m
13:26:08.928837 [96mINIT --- : error starting clang: error running compile: rpc error: code = NotFound desc = Cannot find build path: Rel: can't make C:\Users\Tomek\AppData\Local\Temp\arduino-language-server3532616197 relative to D:\Tomek\Documents\Arduino\LangSrvTest[0m
13:26:08.929034 [92m                 textDocument/didOpen: [93mlocked (waiting clangd)[0m[0m
13:26:08.929563 [92m                 textDocument/didOpen: clangd startup failed: quitting Language server[0m
``` 
[LangSrvTest.zip](https://github.com/arduino/arduino-ide/files/11714474/LangSrvTest.zip)

per1234, 2023-06-11:
>the problem is that the TEMP folder is on another drive, and its path can't be made relative to the Sketch folder

This is a known bug: https://github.com/arduino/arduino-cli/issues/2156

The bug has already been fixed by https://github.com/arduino/arduino-ide/pull/2083 but the version of Arduino IDE you are using doesn't have the fix.

Please try again with the latest nightly build. The download links for the nightly build are listed here:

https://www.arduino.cc/en/software#nightly-builds

The fault you encountered is different from the subject of this issue because https://github.com/arduino/arduino-cli/issues/2156 is a recent regression from after the time of the original report.