arduino/arduino-ide #1044: Duplicate Tabs and path in tabs : Found a cause and remedy 2.0.0 RC7

2ant, 2022-06-11:
### Describe the problem

I opened a project and i noticed that some tabs had the file path in it and that some tabs had duplicates

[image]

I did some searches about it and some people had different cause for it like opening from recent,...

I found the cause in my case : My project is a copy made by the "save as" menu. By saving that way the editor adds "_copy" and the date and time to the saved filename.

By deleting the Date and Time from the .ino file and the folder, my problem disappears.

Maybe the cause is something to do with filename length or "total file path" too long ?

I hope it will help someusers or the Arduino team.

Have a nice day !

### To reproduce

Maybe make a long file name with lots of caracters in it ?

### Expected behavior

Duplicate Tabs and file path in tabs

### Arduino IDE version

2.0.0 RC7

### Operating system

Windows

### Operating system version

Win 10 family 21H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-06-24:
Please try with [`2.0.0-rc8`](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-rc8) and let us know if it still happens. If yes, I will reopen the task and investigate. Thank you!

2ant, 2022-06-24:
@kittaakos  I launched arduino from the desktop icon. It asked me to update and i did.

After restarting arduino, i tried to save my currently open project.

The save happened ( the files are created and my project become the saved one) but the editor has a few problems:

[image]

-I get an empty window with no tabs or board

-When i click on file > open, nothing happens

-When i click on the red cross to quit the program, nothing happens

It seems like the ui is partly functionnal but no action are started after a click ( i had to kill the process to close the IDE.

After closing and opening the file directly from windows

[image]

No more duplicate tabs or long filenames !

I did the same thing a second time with same results:

[image]

[image]

So in conclusion thank you for correcting the "tabs" bug and it would be nice if you could check the "saving make the editor kinda freeze" bug.

kittaakos, 2022-06-24:
> -I get an empty window with no tabs or board
> 
> -When i click on file > open, nothing happens

Thank you! I need some info to further investigate it.

Can you please tell me the path where you installed IDE2? Maybe this is the problem: https://github.com/arduino/arduino-ide/issues/106

How did you install IDE2? Installed or zip? If installer, how did you install it? Maybe there is a permission issue.

If you're willing to help, please start the IDE2 (`"Arduino IDE.exe"`) from a terminal (`CMD.EXE`), and share the logs with us.

2ant, 2022-06-24:
The IDE was installed in the default location i think ( i had the just for me option checked during the update).

[image]

The IDE was installed when after launching the 2.0.0 rc7 i had an "update availiable" message.

I can't launch the IDE from the cmd.exe

[image]

kittaakos, 2022-06-24:
> I can't launch the IDE from the cmd.exe

Please run `"Arduino IDE.exe"`. Note the `"`. Thank you!

2ant, 2022-06-24:
Already tried ?

[image]

2ant, 2022-06-24:
```
Microsoft Windows [version 10.0.19044.1766]
(c) Microsoft Corporation. Tous droits réservés.

C:\Windows\system32>Arduino IDE
'Arduino' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Windows\system32>Arduino IDE.exe
'Arduino' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Windows\system32>cd C:\Users\antoi\AppData\Local\Programs\Arduino IDE

C:\Users\antoi\AppData\Local\Programs\Arduino IDE>"Arduino IDE.exe"

C:\Users\antoi\AppData\Local\Programs\Arduino IDE>
Starting backend process. PID: 12832
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/antoi/.theia'
root INFO Backend Object.initialize: 3.1 ms [Finished 0.887 s after backend start]
root INFO Backend Object.configure: 2.4 ms [Finished 0.887 s after backend start]
root INFO Backend MessagingContribution.onStart: 1.1 ms [Finished 0.887 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 0.924 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.3 ms [Finished 0.924 s after backend start]
root INFO Backend LocalizationBackendContribution.initialize: 37.0 ms [Finished 0.924 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.4 ms [Finished 0.924 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.2 ms [Finished 0.924 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 1.3 ms [Finished 0.926 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 1.2 ms [Finished 0.926 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 1.5 ms [Finished 0.926 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.4 ms [Finished 0.927 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 0.4 ms [Finished 0.927 s after backend start]
daemon INFO Starting daemon from C:\Users\antoi\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
root INFO Backend NodeFileUploadService.configure: 1.4 ms [Finished 0.928 s after backend start]
root INFO Backend PlotterBackendContribution.initialize: 0.2 ms [Finished 0.928 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.2 ms [Finished 0.928 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 0.8 ms [Finished 0.929 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.4 ms [Finished 0.930 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.6 ms [Finished 0.931 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 0.932 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 0.1 ms [Finished 0.932 s after backend start]
root INFO Theia app listening on http://localhost:64342.
root INFO Finished starting backend application: 5.7 ms [Finished 0.935 s after backend start]
root WARN The local plugin referenced by local-dir:/c%3A/Users/antoi/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/antoi/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\antoi\.arduinoProIDE\plugins does not exist.
Restoring workspace roots: c:\Users\antoi\Documents\Arduino\EspNOW_Remote_copy_20220624144759_copy_20220624145133
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-nl" to a VS Code extension "vscode-language-pack-nl@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-tr" to a VS Code extension "vscode-language-pack-tr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... 187 lines cut ...]
[... cut ...]
{"level":"info","lib":"FS","msg":"  found better lib","prio":"64C","time":"2022-06-24T16:07:35+02:00"}

daemon INFO {"level":"info","msg":"Resolving include SPIFFS.h for arch esp32","time":"2022-06-24T16:07:36+02:00"}
{"level":"info","lib":"SPIFFS","msg":"  found better lib","prio":"64C","time":"2022-06-24T16:07:36+02:00"}

daemon INFO {"level":"info","msg":"Resolving include AsyncDelay.h for arch esp32","time":"2022-06-24T16:07:37+02:00"}
{"level":"info","lib":"AsyncDelay","msg":"  found better lib","prio":"643","time":"2022-06-24T16:07:37+02:00"}

```

kittaakos, 2022-06-24:
You're great support. Thank you!

> -I get an empty window with no tabs or board

Please attach a more details screenshot. I need to see the bottom of the IDE2 window. There's the status bar.

Like this:

[image]

Is it purple? If yes, it's the same as https://github.com/arduino/arduino-ide/issues/1089.

> "saving make the editor kinda freeze" bug.

Are you working on remote or local sketches?

> Can you please tell me the path where you installed IDE2? Maybe this is the problem: #106

I can confirm that I do not see any traces of the _"path too long"_ issue in the logs.

2ant, 2022-06-24:
I'm sorry i will keep adding weird issues.

I don't remember if it was purple but it shouldn't have been because : 

1-if it is related to a non existing path ( when i save my project all files are created so path are valid ?)

2-keep reading ( fourth try )

Arduino install path is dafault.

I tried the same thing as earlier : save as... 

weird issue first try:   but this time i got a "hanging screen (files are created normally: 

[image]

I still have the "UI menu work but doesnt open a file"

[image]

but the closing cross work this time !

weird issue second try : everything worked normally (files are created and new copy project is the active project

No idea why everything seem to work this time.

I then erased the multiple copy folders and restarted the IDE.

weird issue third try : it must have remembered the path of the multiple copie since i was greeted by this : 

[image]

I thought "ok, it's normal i just have to open a project.

I did and decided to do the steps AGAIN.

weird issue fourth try : back to square one it seems : 

[image]

No purple like the first times i think.

I noticed something that may help you, some options are lacking : 

[image]

No sketchbook, recent, Examples...

I just remembered i have had this bug sporadically before but not sure why.

I will try again tomorrow or the day after to try to pinpoint the cause more.

kittaakos, 2022-07-26:
@2ant, could you please try if you still have the issues with the nightly build. You can get the nightly from [here](https://github.com/arduino/arduino-ide/#nightly-builds). Thank you!

2ant, 2022-07-27:
@kittaakos  I took a nightly build (2.0.0-rc9-20220727).

I open a project that had already been re saved ( EspNOW_Remote_copy_20220624144759 ), it opened fine.

I tried to save as... and i got the same hanging as before and the files are created (with a different moving circle thingy). 

[image]

I closed the nightly, deleted my "copy copy" and then reopened the nightly and was greeted by the normal empty sketch.

[image]

So for now i just got the "saving get the editor into hanging mode" bug but restarting with a deleted project doesn't f*** up the UI anymore.

I will come back here if i see other related problems in the future.