arduino/arduino-ide #1047: Keyboard Key Number 1 don't work

samtd, 2022-06-11:
### Describe the problem

Key Number 1 don't work.
I try on other programs, like notepad and work.
Also don't work with screen keyboard.

### To reproduce

\234567890'«

### Expected behavior

\1234567890'«

### Arduino IDE version

Version: 2.0.0-rc7 Date: 2022-06-09T08:20:00.114Z CLI Version: 0.23.0 [899dc91b] 

### Operating system

Windows

### Operating system version

Windows 10 Pro 21H2 19044.1706

### Additional context

Keyboard: QWERTY Portuguese

Also ckeck with:
Version: 2.0.0-rc7-nightly-20220611
Date: 2022-06-11T03:26:16.795Z
CLI Version: 0.23.0 [899dc91b]

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-11:
Possibly related to https://github.com/arduino/arduino-ide/issues/989

samtd, 2022-12-14:
I also tested with Theia IDE and works.
Theiablueprint cloud: Work
Theiablueprint installed v1.32.0: Work
arduino-ide_2.0.3_Windows_64bit.zip : Dont Work
arduino-ide_2.0.3_Windows_64bit.exe : Dont Work
Arduino Web App Version - 6.4.6: Work

Edição Windows 11 Pro
Versão 22H2
Instalado em ‎08/‎10/‎2022
Compilação do SO 22621.900

If i use key number 1 with shift it works the correct character "!", but number 1 i can't have
all other keys also works