arduino/arduino-ide #1051: Serial Monitor does not send data to Arduino (2.0.0 rc7)

tigoe, 2022-06-12:
### Describe the problem

Serial Monitor does not send data to Arduino

### To reproduce

1. Open File -> Examples -> Communications -> Physical Pixel
2. Change `ledPin` to `LED_BUILTIN`
3. open serial monitor
4. Type H then enter
5. nothing happens
6. Type L then enter
7. nothing happens
8. Repeat steps 4 - 7 until you are satisfied

For other tests, try File -> Examples -> Communications -> ReadAsciiString. Same results.

All sketches respond just fine in other serial terminals, such as [CoolTerm](https://freeware.the-meiers.org/) or [Serial Studio](https://serial-studio.github.io/)

### Expected behavior

The LED should turn on with H and off with L. With the ReadAsciiString sketch, I should get three values back in hexadecimal. 

### Arduino IDE version

2.0.0-rc7

### Operating system

macOS

### Operating system version

12.4 (21F79)

### Additional context

Same results on Windows 11.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-13:
Hi @tigoe

>Type H then enter

Unlike the Arduino IDE 1.x Serial Monitor, it is currently necessary to use <kbd>Ctrl</kbd>+<kbd>Enter</kbd> rather than just <kbd>Enter</kbd> alone in order to send data to the board from Serial Monitor (https://github.com/arduino/arduino-ide/issues/572).

Please give that a try and then let me know whether the issue still occurs for you.

PaulStoffregen, 2022-06-13:
fwiw, I tried it on Linux with Arduino Zero and Teensy 4.1 (to also check whether any difference exists between Teensy's pluggable monitor and Arduino's implementation).

Even having just read this, my fingers *still* didn't hold the CTRL key on the first few tries!  So awkward.

Can confirm, both do change their LED running the PhysicalPixel example when I use CTRL+Enter to send.

PaulStoffregen, 2022-06-13:
Slightly off-topic, 04.Communication > PhysicalPixel has pin 13 hard coded.  Might be good to update it to LED_BUILTIN for compatibility with Arduino's newer products...

ubidefeo, 2022-06-13:
@PaulStoffregen 
I have made a PR for that `LED_BUILTIN`

per1234, 2022-06-13:
Thanks everyone. Since we now know that the issue was caused by https://github.com/arduino/arduino-ide/issues/572, I will close this issue in favor of the other one.

closed 2022-06-17 by per1234