arduino/arduino-ide #1054: "Go to Definition" results in duplicate tabs when target is in sketch created in previous IDE version

per1234, 2022-06-13:
### Describe the problem

The Arduino IDE's language server-based "Go to definition" feature reveals the definition of a code component. If that definition is in a file that is not already open in an editor tab, a tab must be added for that file. If the file is already open in a tab then that tab is simply selected and the cursor placed at the start of the definition.

🐛 If the sketch was created in an earlier version of the Arduino IDE, "Go to definition" for a definition within one of the open sketch editor tabs causes the creation of a confusing duplicate tab instead of simply using the existing tab.

### To reproduce

1. Start Arduino IDE 2.0.0-rc6.
1. Select **File > New** from the Arduino IDE menus.
1. Create a sketch that contains a function definition and a call to that function:
   ```cpp
   void setup() {
     foo();
   }
   void loop() {}
   void foo() {}
   ```
1. Save the sketch to any convenient name and location.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the latest version of the Arduino IDE.
1. Open the sketch you created if it is not already.
1. Select any board with language server support (e.g., Arduino Uno) from the **Tools > Board** menu in the Arduino IDE.
1. Right click on the function call:
   ```cpp
   foo();
   ```
1. Select "**Go to Definition**" from the context menu.

🐛 A duplicate tab is opened.

🐛 The cursor is not moved to the start of the function definition.

[image]

### Expected behavior

Duplicate tabs are not created.

The cursor is moved to the start of the function definition.

### Arduino IDE version

2.0.0-rc7

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

Related:

- https://github.com/arduino/arduino-ide/issues/1044
- https://github.com/arduino/arduino-ide/pull/1024
- https://github.com/arduino/arduino-ide/issues/656
- https://github.com/arduino/arduino-ide/issues/793

---

Originally reported at https://forum.arduino.cc/t/ctrl-click-navigation/1000117/19

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-06-17:
Crazy, but this is the problem: when opening an editor, all widget open `options` must be the same. See [here](https://github.com/arduino/arduino-ide/pull/1012#issue-1252816253).

> (Since editor widget identity is based not only on the URI, but also the options, we need to pass the correct options into the open call. Doing that prevents Theia from opening multiple tabs of the same editor)

It means, when opening `/path/to/resource/MySketch/MySketch.ino` file in the editor, the `uri`, the `counter`, `kind`, .etc must be the same.

But the order of the properties also matters 🤦

This is the known, opened editor widget in the widget manager:
```json
{
    "factoryId": "code-editor-opener",
    "options": {
        "uri": "file:///Users/a.kitta/Documents/Arduino/xxx2/xxx2.ino",
        "kind": "navigatable",
        "counter": 0
    }
}
```

This is the editor IDE2 wants to open with `Go to Definition`:
```json
{
    "factoryId": "code-editor-opener",
    "options": {
        "kind": "navigatable",
        "uri": "file:///Users/a.kitta/Documents/Arduino/xxx2/xxx2.ino",
        "counter": 0
    }
}
```

The two objects are deeply equal, but they're not the same. [Here](https://github.com/eclipse-theia/theia/blob/024d8714393cc0d8f821d6350cba71c6df4a63ad/packages/core/src/browser/widget-manager.ts#L195) is the corresponding code in Theia.

kittaakos, 2022-06-17:
I can confirm that I could reproduce similar defect with vanilla Theia: https://github.com/eclipse-theia/theia/issues/11309