arduino/arduino-ide #1055: Use dedicated location for "built-in" libraries

per1234, 2022-06-13:
### Describe the request

Use a dedicated location for the "built-in" library installations, with a lower location priority than [platform bundled libraries](https://arduino.github.io/arduino-cli/latest/platform-specification/#platform-bundled-libraries).

🙂 Library dependency resolution will match the behavior of Arduino IDE 1.x

### Describe the current behavior

In order to allow users to get started with Arduino with the minimum amount of complexity, the Arduino IDE 1.x installation comes with [a collection](https://github.com/arduino/Arduino/blob/1.8.19/build/build.xml#L254-L274) of fundamental Arduino libraries (e.g., [Keyboard](https://github.com/arduino-libraries/Keyboard), [Mouse](https://github.com/arduino-libraries/Mouse), [Servo](https://github.com/arduino-libraries/Servo)) pre-installed. Even experienced users now expect these libraries to be available by default, and many tutorials (both official and 3rd party) have been written based on that assumption. So it is important that Arduino IDE 2.x also provide these libraries.

In order to simplify the Arduino IDE packaging process and ensure users are provided with the newest available, the Arduino Library Manager infrastructure is currently used to install the libraries the first time the Arduino IDE is started (https://github.com/arduino/arduino-ide/pull/663).

Library Manager installs these libraries to the sketchbook folder. Libraries at this location have a higher ["location priority"](https://arduino.github.io/arduino-cli/latest/sketch-build-process/#location-priority) than they did when bundled with the Arduino IDE installation.

🙁 This priority increase can cause the "built-in" library to now be selected by the Arduino sketch build process in a case where a platform bundled library would have been selected when using Arduino IDE 1.x.

### Arduino IDE version

2.0.0-rc7

### Operating system

All

### Operating system version

N/A

### Additional context

This would also provide an alternative solution to https://github.com/arduino/arduino-ide/issues/798

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

ubidefeo, 2022-06-14:
@per1234 
it seems we thought we were solving an issue by bundling the legacy libraries but we caused another one 🤔

kittaakos, 2022-08-30:
Upstream PR: https://github.com/arduino/arduino-cli/pull/1817

commit fb588a44 (2022-08-31): Install the `Arduino_BuiltIn` to built-in location

Closes #1055.
PR #1370 (2022-09-02): #1055 Install the `Arduino_BuiltIn` to built-in location
### Motivation

Use a dedicated install location for the `Arduino_BuiltIn` lib on the first IDE2 start.

### Change description
From now on, the built-in lib will be installed into the `built-in` location of the CLI.

Closes #1055.

### Other information

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / update [... cut]
commit 1b7f86b2 (2022-09-02): Install the `Arduino_BuiltIn` to built-in location

Closes #1055.
closed 2022-09-02 by kittaakos