arduino/arduino-ide #1059: Control+Click not working

qwertychouskie, 2022-06-13:
### Describe the problem

When trying to use Ctl+Click on a function/variable/etc, it doesn't work.  It will sometimes work when opening one of the example programs, but never on my own program.

### To reproduce

Control+Click

### Expected behavior

Control+Click should work reliably

### Arduino IDE version

2.0.0-rc7 (via Flatpak)

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

qwertychouskie, 2022-06-13:
Things seem to be working more consistently after messing with a few things, including enabling "Editor Quick Suggestions" in the settings, I'll see if it continues to work or starts bugging out again.

per1234, 2022-06-14:
Thanks for your report @qwertychouskie.

There was a previous report of this problem on the Arduino forum. After some investigation, it was discovered that the cause was https://github.com/arduino/arduino-ide/issues/1054

>The cursor is only moved to the start of the function definition in the newly created tab, and that tab is not selected, so the user is not guided to the function definition in any meaningful way.

If you encounter this problem again, please check to see whether the Arduino IDE shows two copies of the sketch tag, similar to what is shown in this screenshot:

[image]

qwertychouskie, 2022-06-14:
That is a different issue, in my case Ctl+Click wouldn't do anything at all.

joshuag1000, 2022-06-15:
Like i have just discovered in an issue i had open (#1041) where the setting dialog didnt work this issue could be related to the faster loading that was implemented in rc7. I also had this issue until i let the ide sit for a minute and load all the libraries etc in the background and then the ctrl + click was working again.
If you let the ide load for a few minutes does ctrl + click work again?

per1234, 2022-07-13:
From https://github.com/arduino/arduino-ide/issues/714#issuecomment-1158237618 it appears that @qwertychouskie was able to determine the problem is the same as https://github.com/arduino/arduino-ide/issues/714.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-07-13 by per1234

Banderi, 2023-08-22:
> From [#714 (comment)](https://github.com/arduino/arduino-ide/issues/714#issuecomment-1158237618) it appears that @qwertychouskie was able to determine the problem is the same as #714.
> 
> It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.
> 
> If you end up with additional information to share, feel free to comment in the other thread.

I tried reading through #714 but I can't seem to find what is the connection with this issue, or what/if there is any solution to it.
This issue sadly still happens on my end, on 2.1.1 exactly as described above.

Was there a way to fix this?

wojtron2, 2024-02-16:
Same issue appeared on 2.3.0 here