arduino/arduino-ide #1063: CPU and memory usage never stop increasing

marc0777, 2022-06-15:
### Describe the problem

CPU and memory usage continuously increase while using the IDE. CPU usage seems to settle at around 25% and never go over, while memory usage doesn't seem to have an upper limit. 

Moreover memory usage varies a lot by the second:

[image]
:

### To reproduce

While I wasn't able to find a specific triggering event, this doesn't seem to be related to building or uploading, opening new tabs or any other single action. 

Instead, it resource usage continuously increase as I edit the code. Keeping the IDE open while just reading the code, compiling and uploading doesn't trigger this behaviour, but after some minutes writing code the IDE starts to feel slower and slower, as resource usage grows.

### Expected behavior

Resource usage should only be noticeable during initial indexing, compilation and upload. Instead it should be very low while editing the code itself.  

### Arduino IDE version

nightly-20220615

### Operating system

Windows

### Operating system version

11 21H2

### Additional context

This issue is been present since at least rc6, and is present on rc7 as well. 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

marc0777, 2022-06-15:
Further searching into the issues I found #714, that describes the same problem I'm encountering. This makes this IDE almost impossible to really use, since on a laptop it drains the battery very quickly while warming the PC considerably...

qwertychouskie, 2022-06-16:
Definitely looks like https://github.com/arduino/arduino-ide/issues/714, can probably close this one as a duplicate.

per1234, 2022-09-22:
Hi @marc0777. Thanks for your report. #714 was resolved some time ago. Please try the latest version of the Arduino IDE and then comment here to tell us whether the problem still occurs.

The download links are listed here:

https://www.arduino.cc/en/software

marc0777, 2022-10-11:
Hello, in the last few days I had the chance to test the new version and the issue seems to be completely solved!

closed 2022-10-11 by per1234