arduino/arduino-ide #1072: Teensy Plugable Serial Monitor Working with new daily builds?

KurtE, 2022-06-17:
### Describe the problem

Originally I had some comments under: #1051  which was closed today

But Actually it appears to be  a little more involved than I first thought.  And probably should be it's own track-able issue. 
@per1234 @PaulStoffregen - not sure which component this be in?
@mjs513 - not sure if you also saw something similar when running on RPI with Ubuntu?

Example sketch:

```
elapsedMillis em = 0;
uint32_t alive_count = 0;
void setup() {
  pinMode(13, OUTPUT);
  while (!Serial)
    ;
  Serial.begin(115200);
  delay(1000);
  Serial.println("Start of Echo Test");
}

void loop() {
  if (Serial.available()) {
    digitalToggleFast(13);
    Serial.print("Received:(");
    int ch;
    while ((ch = Serial.read()) != -1) Serial.write(ch);
    Serial.println(")");
  }
  if (em > 5000) {
    // If I Uncomment the print below it echos???
    //Serial.printf("Alive Count: %u\n", ++alive_count);
    em = 0;
  }
}
```

Without the Serial.printf for alive count, the 

### To reproduce

Using something like the sketch included

I am using today's daily build, with Teensyduino installed 
using URL: https://www.pjrc.com/teensy/td_156/package_teensy_index.json
with current release actually 1.57 Beta 3

I build for Teensy 4.1, USB Type: Serial
Although I saw it on yesterdays build with other USB types as well.

So if I build the above sketch: and run it with Teensy Port selected
[image]

Typing something in the text area and hitting CTRL+enter does not echo any of the txt nor change state of LED...
But if I then change to COM27 instead:
[image]

It does echo and led changes state.

Now the interesting point is now if I uncomment the line that outputs alive count...
Then it appears to receive input from the Serial port... But maybe only after it writes something to the Serial Port
[image]
That is there can be a several second delay (i.e. my 5 second elapsedMillis .

But running on COM27 there is no such delay

### Expected behavior

Expect it to work the same, and should receive the data from Serial as soon as the user sends it. 

### Arduino IDE version

Nightly 20220617

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

KurtE, 2022-06-17:
Notes:
@PaulStoffregen ran it on Ubuntu and simple sketch worked fine (did not reproduce the issue)
@mjs513 and myself ran it on RPI4 running Ubuntu (RC7) and it worked fine
I also tried it on MAC using daily build and it worked find. 

So it looks like Windows Specific thing

per1234, 2022-06-18:
Thanks @KurtE. I am also able to reproduce it with [`arduino-cli monitor`](https://arduino.github.io/arduino-cli/dev/commands/arduino-cli_monitor/):

```text
$ arduino-cli version
arduino-cli.exe  Version: git-snapshot Commit: 64bc4dce Date: 2022-06-17T13:21:10Z

$ arduino-cli monitor --fqbn teensy:avr:teensy41 --port usb:0/1A0000/0/1/5/1/1/1 --protocol teensy
Connected to usb:0/1A0000/0/1/5/1/1/1! Press CTRL-C to exit.
Start of Echo Test
asdf
```

```text
$ arduino-cli monitor --fqbn teensy:avr:teensy41 --port COM40 --protocol serial
Connected to COM40! Press CTRL-C to exit.
Start of Echo Test
asdf
Received:(asdf
)
```

And Arduino CLI also behaves as reported after uncommenting the "Alive Count" output line:

```text
$ ./arduino-cli monitor --fqbn teensy:avr:teensy41 --port usb:0/1A0000/0/1/5/1/1/1 --protocol teensy
Connected to usb:0/1A0000/0/1/5/1/1/1! Press CTRL-C to exit.
Start of Echo Test
Alive Count: 1
Alive Count: 2
asdf
Alive Count: 3
Received:(asdf
)
Alive Count: 4
```

This tells me that the issue is not in the Arduino IDE code base. However, I'm not sure whether it is in Arduino CLI or in teensy-monitor. I'll transfer this to the Arduino CLI repo for now as that is at least closer to the correct place even if possibly not exactly it.