arduino/arduino-ide #1073: Last lines of compilation output are not printed

per1234, 2022-06-17:
### Describe the problem

🐛 The last lines of compilation output are not shown in Output panel

### To reproduce

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Uncheck the box next to **Show verbose output during > `☑` compile**.
   **ⓘ** The issue occurs with the verbose output as well. This is done only to produce more simple output for the demo.
1. Click the <kbd>**OK**</kbd> button.
1. Select **Tools > Board > Arduino AVR Boards > Arduino Uno** from the Arduino IDE menus.
   **ⓘ** The issue is not specific to this board. A board selection is specified only to ensure the output described here in the demo will be consistent with that seen by those who are reproducing it.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation process to finish.
1. If the issue does not occur, repeat the process from step 5 until it does occur.

🐛 The memory usage output is not shown:

```text

--------------------------
Compilation complete.
```

The output should be:

```text
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

--------------------------
Compilation complete.
```

### Expected behavior

All lines of compilation output

### Arduino IDE version

2.0.0-rc7-snapshot-94ceefd

### Operating system

Windows, Linux

### Operating system version

Windows 10, Ubuntu 20.04

### Additional context

I bisected the issue to https://github.com/arduino/arduino-ide/commit/4c6243176ca5e66635fde4107e217f1b9666eb62 (does not occur when using the build for https://github.com/arduino/arduino-ide/commit/a8047660a6e23fa41bc42457cf27e53bfe5798d8).

---

The number of missing lines varies. This is easier to observe with verbose compilation output enabled. Sometimes the line showing the `avr-size` command that would be expected immediately before the memory usage lines is printed:

```text
[...]

"C:\\Users\\per\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\per\\AppData\\Local\\Temp\\arduino-sketch-13926962AEE9C72B0EB120253E6668CD/sketch_jun17a.ino.elf" "C:\\Users\\per\\AppData\\Local\\Temp\\arduino-sketch-13926962AEE9C72B0EB120253E6668CD/sketch_jun17a.ino.hex"

"C:\\Users\\per\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "C:\\Users\\per\\AppData\\Local\\Temp\\arduino-sketch-13926962AEE9C72B0EB120253E6668CD/sketch_jun17a.ino.elf"

--------------------------
Compilation complete.
```

Other times it is not:

```text
[...]

"C:\\Users\\per\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\per\\AppData\\Local\\Temp\\arduino-sketch-13926962AEE9C72B0EB120253E6668CD/sketch_jun17a.ino.elf" "C:\\Users\\per\\AppData\\Local\\Temp\\arduino-sketch-13926962AEE9C72B0EB120253E6668CD/sketch_jun17a.ino.hex"

--------------------------
Compilation complete.
```

More rarely, one or two of the memory usage lines are shown.

---

Originally reported at https://forum.arduino.cc/t/memory-usage-display-after-compile/1003909

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-27:
A report on the forum revealed that the problem still occurs:

https://forum.arduino.cc/t/2-0-rc8-nightly-20220627-ram-usage-not-displayed/1007235

I can reproduce this. What I observe is that, even using the build from the time of the fix (https://github.com/arduino/arduino-ide/commit/a715da3d1839ee7d0721af13515581101dae785f), the last line of the compilation output:

```text
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
```

is sometimes not printed, but the problem occurs very infrequently (approximately 3% of compilations). Something changed at https://github.com/arduino/arduino-ide/commit/a54d7c8f45a1348b25798ce117f287af9ee17a5d. Using that build and all after that (up to the current state at 2.0.0-rc8), it is much more frequent (approximately 30% of compilations).

davegarthsimpson, 2022-06-28:
likely related to https://github.com/arduino/arduino-ide/pull/1064 which was merged into https://github.com/arduino/arduino-ide/pull/1040 from main (this was the last time related code was modified)

davegarthsimpson, 2022-06-28:
This appears CLI related, when the last line is cut off it is not received at all by the IDE;

I actually get the line cut off around 30-70% of the time with the latest CLI.

I get around 1-3% if I use a build of the CLI **omitting** the change in [this PR](https://github.com/arduino/arduino-cli/pull/1759)

per1234, 2022-07-01:
Since it turned out to be a bug in Arduino CLI, I'm going to close this as resolved by the fix: https://github.com/arduino/arduino-cli/pull/1787

The IDE will continue to produce the fault, but that is only because it is still using the version of Arduino CLI that had the bug. The fix will propagate to Arduino IDE as a matter of course on the next update of the Arduino CLI dependency, without any special action required in this code base.

Thanks Dave for your excellent work on identifying the source of the fault.

closed 2022-07-01 by per1234