arduino/arduino-ide #1076: Splash screen can open on different screen then app

boundmaidlea, 2022-06-19:
### Describe the problem

When opening the app from e.g start menu on one screen, the splash screen will always display on this screen, unlike the app which opens on the screen it was closed on last session. This, combined with #324 and #193 potentially denys all multitasking while the IDE loads, which also takes pretty long. 

### To reproduce

* Use 2 screens
* Open the IDE
* Note which screen the IDE is on
* Close it again
* Use the start menu on the other screen to open the IDE

### Expected behavior

The Splash screen opens on the same screen the IDE opens on, and ideally even follows the IDE if it were to be moved to another screen.

### Arduino IDE version

2.0.0-rc7-nightly-20220615

### Operating system

Windows

### Operating system version

10.0.19044.1706 Pro

### Additional context

My Nightly Build is 4 days old, but the 3 commits since then do not have any relations to this issue.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

Defragster, 2022-07-21:
Splash seems to activate on the 'last clicked/Active' screen - as seen here on Win 11.

About to open an issue that IDE 2 won't start here - but start of EXE may have a desired screen, but click the other screen before it appears and that is where I see the Splash appear.

per1234, 2022-09-22:
Closing as resolved by the removal of the splash screen completely in https://github.com/arduino/arduino-ide/pull/1287

closed 2022-09-22 by per1234

Defragster, 2022-09-22:
Indeed, as released without separate splash screen - Nice.