arduino/arduino-ide #1077: Some Theia settings have no effect, problematic decriptions, or are ignored on startup

boundmaidlea, 2022-06-19:
### Describe the problem

Settings like 
* editor.padding.bottom
* editor.padding.top
* window.menuBarVisibility
* window.titleBarStyle

have no effect at all, even though it would be desirable to have these settings changable, and `breadcrumbs.enabled` only works in specific circumstances, but otherwise doesn't fulfill any actual purpose atm (showing the file's path), even though it could.

---

The setting `editor.minimap.scale` will be saved in the settings.json file, but is not read on startup and needs to be re-set in the settings to take effect.

---

Further the description of the `editor.language.colorizedBracketPairs` setting's description states

> Use 'workbench.colorCustomizations' to override the bracket highlight colors.

Even though `workbench.colorCustomizations` is not supported by this IDE at the moment.

---

The description of `application.confirmExit` seems to assume that the app runs in browser even if it doesn't.

### To reproduce

* open the command palette
* use the `Preferences: Open Settings (UI)` command
* change or look at one of the mentioned settings

### Expected behavior

all mentioned settings do what they are described to do and vice versa.

### Arduino IDE version

2.0.0-rc7-nightly-20220615

### Operating system

Windows

### Operating system version

10.0.19044.1706 Pro

### Additional context

My Nightly Build is 4 days old, but the 3 commits since then do not have any relations to this issue.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-29:
This issue was partially resolved by https://github.com/arduino/arduino-ide/pull/1662

The remaining unresolved components:

- `breadcrumbs.enabled` has no effect
- `window.titleBarStyle` has no effect
- `workbench.editor.highlightModifiedTabs` has no effect

per1234, 2022-12-01:
As I explained above, most of the reported problems were fixed upstream in Theia and those fixes have now been pulled into Arduino IDE via https://github.com/arduino/arduino-ide/pull/1662. A new issue has been created to track the remaining problems reported here: https://github.com/arduino/arduino-ide/issues/1733

So I will close this as resolved. Thanks @boundmaidlea!

closed 2022-12-01 by per1234