arduino/arduino-ide #1079: Allow opening Examples in same window.

boundmaidlea, 2022-06-19:
### Describe the request

When intending to open one of the examples or another previous project's file to take reference, it would be great if it is possible to do so in the same window to use split screen view. To prevent issues with different project configuration/etc. such files could be made read-only.

### Describe the current behavior

Currently each opened example and all files opened through the file>"open" option opens an entirely new instance of the IDE.

### Arduino IDE version

2.0.0-rc7-nightly-20220615

### Operating system

Windows

### Operating system version

10.0.19044.1706 Pro

### Additional context

My Nightly Build is 4 days old, but the 3 commits since then do not have any relations to this feature.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

boundmaidlea, 2022-06-19:
Seems to be related to #114

per1234, 2022-06-23:
Thanks for your request @boundmaidlea. The Arduino IDE is intentionally designed to only allow a single sketch to be opened in each window. The reason is that a sketch may consist of multiple files, which are all compiled as a single program. It is important to intuitively communicate that fact to the users, which would no longer happen if they had multiple sketches open in the same window.

I don't think the requirement of having multiple Arduino IDE windows is onerous.

boundmaidlea, 2022-06-23:
>  It is important to intuitively communicate that fact to the users

I do understand that, that is another reason why i meant it would make sense to open them in read-only mode.
I can hardly see new users being confused, if allowing this requires to switch a setting first
Further i could see a special coloured border/etc. being used. 

The app could also still open outside files in a new instance, but being able to drag&drop a tab into another instance would already be a great improvement in workflow for me, as that would already allow me to use split view

One of the main issues with multiple instances for me is, that the performance is already bad enough with it running one time, and having it opened a second time actually starts to lag out my whole system after some time.

boundmaidlea, 2022-06-23:
Oh i didn't even see the Editor split feature being removed until seeing this: https://github.com/arduino/arduino-ide/issues/715#issuecomment-1163780614_ because i didn't even get that far, that's disappointing...

djcarter, 2023-03-07:
> I don't think the requirement of having multiple Arduino IDE windows is onerous.

You are wrong.