arduino/arduino-ide #1080: Add setting for Panel side

boundmaidlea, 2022-06-19:
### Describe the request

Have a setting similar to VSCode's `workbench.panel.defaultLocation` setting, that changes the side of the Panel in which the Output and Terminal tabs will open. Possible values will be the same as for `workbench.panel.defaultLocation`:
* left
* right
* bottom (default, same as current)

### Describe the current behavior

Currently the Panel will be at the bottom only, which is not ideal for widescreen users.

### Arduino IDE version

2.0.0-rc7-nightly-20220615

### Operating system

Windows

### Operating system version

10.0.19044.1706 Pro

### Additional context

My Nightly Build is 4 days old, but the 3 commits since then do not have any relations to this feature.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

boundmaidlea, 2022-06-19:
Just looked for duplicates for another issue, and found #715 to be related

toogooda, 2022-07-20:
The old version was at least able to hide the output panel, the new implementation with no ability to move or hide is a large regression from the original arduino IDE vertical space is much less than horizontal for 99% of all users. Options to move or hide are a minimum.