arduino/arduino-ide #1081: Serial monitor selection range shifts chaotically upon scroll

boundmaidlea, 2022-06-19:
### Describe the problem

When selecting text on the serial monitor (with the connected device paused) upon scroll the range of the selection shifts in unpredictable patterns (at least i didn't see a reproducable pattern) which prevents selecting more then one page of the output. As of #812/#105 using ctrl+a is also not an option to select text in that tab.

As i do not have the problem of new text triggering a redraw which removes the selection, i do not consider this a duplicate of #812/#105, but feel free to correct me.

### To reproduce

* print multiple pages text on the serial monitor once e.g. in the setup() function.
* when all text is received select a bit of it
* scroll the output up and down a bit

### Expected behavior

The selection should stay between the first and last selected characters, and not randomly jump around.

### Arduino IDE version

2.0.0-rc7-nightly-20220615

### Operating system

Windows

### Operating system version

10.0.19044.1706 Pro

### Additional context

My Nightly Build is 4 days old, but the 3 commits since then do not have any relations to this issue.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

boundmaidlea, 2022-06-19:
I also just found #247 which is similar but seems unrelated

KurtE, 2022-07-07:
I have been noticing the same thing as well, as when it scrolls, it often only appears to only copy the stuff that is visible on the screen at the time... Have not completely verified that exactly, but that is how it feels.

I was trying to decide if I should append that to my previous issue #812 but saw this one which was more specific

InstantMuffin, 2022-08-12:
I too noticed the various issues mentioned and referred to here.
Features aside, which are nice, the things I would consider bugs (anything around selecting and copying the output) are quite limiting at least my experience with the serial monitor and its possible use cases. Right now it is only useful for slow, continuous output. Anything you want to analyze more or even dump is quite bothersome with the serial monitor as it now.
One could dump this using an external program, however at least on linux the serial device can only be used by a singular process. So you are forced to close the IDE entirely.

aarush3002, 2022-10-22:
Experiencing the same problem - I have a lot of data printed to the serial monitor. When I highlight all of it and try to copy paste, it only copy pastes the last few lines of the output. Have no idea what's going on

Jeff-Haas, 2023-01-18:
Is there any update on the issue of copying text from the Serial Monitor?  I'm trying to decode an IR remote control, and there is a lot of info that I can't copy to a text editor, because only what's visible on the screen gets copied.

chonuchan, 2023-01-25:
I am surprised that this issue is classified as "[type: imperfection]" only. Actually this is a big issue as one cannot collect the data that comes in from serial monitor.

Mishy1973, 2023-01-26:
I would say not only a possibility to select, is needed, but that would be really basic function to get a <Save Output> button for instant Save. Either by predefined format (like "%Y%m%d_%H%M%S.$projectname.log" or whatever) into a preset folder. Or if it is complicated to implement, simply open a FileOpen/Save dialog for the user.

per1234, 2023-01-26:
>I am surprised that this issue is classified as "[type: imperfection]" only.

You misunderstand the meaning of the label @chonuchan. The "**type: imperfection**" label identifies items related to any defect whatsoever. The name was intentionally chosen to be applicable to any component of the project, at any severity on the spectrum from "missing comma in the readme" to "Skynet is waging war on humanity".

The label does not in any way indicate level of severity. We have a separate set of "**criticality**" labels for that purpose. "**criticality**" labels are assigned by a manager during the planning process, while "**type**" labels are assigned by a maintainer during triage.

> get a button for instant Save. Either by predefined format (like "%Y%m%d_%H%M%S.$projectname.log" or whatever) into a preset folder. Or if it is complicated to implement, simply open a FileOpen/Save dialog for the user.

We have a dedicated issue for that proposal: https://github.com/arduino/arduino-ide/issues/549

---

Please, everyone, this is getting noisy and off topic. Please only comment if you have some significant information to add that is directly relevant to the resolution of the issue.

If you only want to express your support, please use GitHub's ["Reactions" feature](https://docs.github.com/en/rest/reactions) to add a thumbs up on the issue or comment you agree with.

Eheran1, 2023-08-23:
> Please, everyone, this is getting noisy and off topic. 

I has been another 8 months. No noise, no off topic, no updates, no fixes, no reversion to the way it worked in 1.x

It used to be a good serial monitor, now it is essentially useless for more than a year. Are you going to look at this? Is this actually something you work on? Can you give us feedback?