arduino/arduino-ide #1083: arduino IDE not compiling after updating to rc7

gargoyle777, 2022-06-20:
### Describe the problem

After updating the IDE, the output monitor disappeared when i tried to create a new project, but was still on screen when opening old project.
I thought i just messed up, so i just used an old project and copy pasyed what i needed compiled, now the problem is that it doesn't compile or upload at all. i've tried to reinstall but im still having the same exact problem.
Other then no output monitor and no compiling, the oard manager and the library manager are not working too, not displaying anything at all.
As of now, i downloaded the 1.x version, and im forced to use that on my project.
working on windows 10

### To reproduce

I just updated the arduino IDE from the pop up on opening it

### Expected behavior

Display board and library on the manager, display the output monitor, verify and upload working

### Arduino IDE version

arduino ide 2.0.0 rc7 windows 64 bit

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-20:
Hi @gargoyle777. We only accept reports of problems that have been verified to occur when using the latest nightly build of the Arduino IDE. Some very significant amount of development has happened since the 2.0.0-rc7 version of the Arduino IDE you are using, including a fix for what appears to be the issue you are reporting: https://github.com/arduino/arduino-ide/issues/1036

Please try again using the latest nightly build and then let me know whether or not the problem still occurs. The nightly build download links are listed here:

https://www.arduino.cc/en/software#nightly-builds

gargoyle777, 2022-06-20:
hi @per1234 ,
thanks for the help, in fact i could chose the board, but i think it was the same issue.
I followed the fix pointed in https://github.com/arduino/arduino-ide/issues/1036 and now it's working again.
Thanks again for your work, i didnt thought about looking for the managers' bug instead of the compile/error's bug