arduino/arduino-ide #1086: Monitor Baud rate 1000000 show giberish

xinkiknix, 2022-06-20:
### Describe the problem

When baud rate is set to 1000000 both in monitor as in application, the show information is not useable.
460800 works fine. I did not test the full range :-(
PC windows 11 build 22000
i7-8700 @3.20Ghz
8GB RAM
SSD

### To reproduce

Set Baud rate to 1000000

### Expected behavior

Shwoing standard ASCII set

### Arduino IDE version

2.0.0 rc7

### Operating system

Windows

### Operating system version

11

### Additional context

PC windows 11 build 22000
i7-8700 @3.20Ghz
8GB RAM
SSD

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-22:
Thanks for your report @xinkiknix.

Please tell me which Arduino board you are using.

Please provide a sketch I can use to reproduce the problem.

xinkiknix, 2022-06-23:
Yes teh ESP32 has a CP2102 module,  the ESP32-S3 has CP2102N version. both show same behaviour