arduino/arduino-ide #1087: Use a more readable color for "Output" view stderr text in the "Light (Arduino)" theme

per1234, 2022-06-21:
### Describe the request

Use a brighter red/orange color for the stderr text printed in the "**Output**" view when using the "**Light (Arduino)**" theme.

I suggest using the Arduino IDE 1.x coloration as a reference, since this has already been user tested for years with favorable opinions (with the exception of those with color blindness, for whom a specialized theme will be more suitable anyway). It is defined here:

https://github.com/arduino/Arduino/blob/43b0818f7fa8073301db1b80ac832b7b7596b828/build/shared/lib/theme/theme.txt#L32

```text
console.error.color = #E34C00
```

🙂 The "**Output**" view will be easier and more enjoyable to read.

### Describe the current behavior

During the compilation or upload operations, output from the processes may be printed to the "**Output**" view. In order to make it easier to differentiate the types of information conveyed there, output that was written to the standard error stream is colored differently from the output written to the standard output stream (https://github.com/arduino/arduino-ide/pull/1064).

The color used for stderr text when the default "**Light (Arduino)**" Arduino IDE theme is active is a fairly dark red color.

Some years ago, this coloration in the Arduino IDE 1.x theme was changed from a brighter color to a dark red very similar to the one currently in use by Arduino IDE 2.x's "**Light (Arduino)**" theme (https://github.com/arduino/Arduino/commit/501333a8ef1fbaf5438fe1613003d146bcc697a8). That resulted in complaints from the users:

- https://github.com/arduino/Arduino/issues/2901
- https://github.com/arduino/Arduino/issues/3056

Until the change was reverted (https://github.com/arduino/Arduino/commit/8020c0d7339ede208f4a60934e65efb9a52b63df). So I think we already have clear evidence that the current coloration in the Arduino IDE 2.x "**Light (Arduino)**" theme is not ideal.

Arduino IDE 2.x:

[image]

Arduino IDE 1.8.19:

[image]

🙁 This very important text is difficult and unpleasant to read.

### Arduino IDE version

2.0.0-rc7-snapshot-4611381

### Operating system

Windows

### Operating system version

10

### Additional context

See https://github.com/arduino/arduino-ide/pull/1064#issuecomment-1157293078 for more information on the subject.

---

With the exception of "**Light (Theia)**" (which has a white "Output" view background), the other themes use a brighter color.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

ObviousInRetrospect, 2022-08-11:
+1 on this ... and is there a way for users to override it in the interim? the error text is really hard to read. I looked for a theme.txt in either /Applications/Arduino IDE or ~/Library/Arduino15 and came up empty.

Tuczz, 2022-09-16:
Hello!

The cursor bar in the Dark theme is too dark too. It's important to be clear where in the code you will write. The pure white would be better in my opinion.

Thanks,
Lucas

per1234, 2022-09-17:
@ObviousInRetrospect please ask over on the Arduino forum about customizing the Arduino IDE theme. I'm sure we'll be able to help you out over there.

https://forum.arduino.cc/c/software/arduino-ide-2-0/93

CoolByDesign, 2023-01-26:
the output text is unreadable for me. I have to highlight the field and even then it is difficult to read. 
For me this issue is serious enough that I cannot use this version.

aoloe, 2023-01-30:
on top of the error text being unreadable, the light theme should provide a light output terminal, not a dark one.

91volt, 2023-02-07:
PR solving the issue: 
https://github.com/arduino/arduino-ide/pull/1836

closed 2023-02-15 by 91volt