arduino/arduino-ide #1110: Boards list widget search must be deterministic

kittaakos, 2022-06-24:
### Describe the problem

The _Boards included in this package_ content is generated by the IDE2 from the result from the CLI. It's not deterministic.

### To reproduce

 - Open IDE2,
 - Open Boards Manager,
 - Search for `esp`,
 - Delete `p` and add `p` so that search runs again with the `esp` term. Now the result is different.

https://user-images.githubusercontent.com/1405703/175551342-270fd4a7-fb85-48c8-a93d-4cdea0c5deba.mp4

### Expected behavior

It's the same.

### Arduino IDE version

rc8

### Operating system

macOS

### Operating system version

12.3.1

### Additional context

Let's order the list of boards alphabetically in the IDE2.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

AlbyIanna, 2022-08-11:
@per1234 it appears to me that the order of the boards that is defined in the [boards.txt configuration file](https://arduino.github.io/arduino-cli/dev/platform-specification/#boardstxt) is the same as the order defined in the [package index](https://arduino.github.io/arduino-cli/dev/package_index_json-specification/#platforms-definitions). Can you confirm it?

Because issue https://github.com/arduino/arduino-ide/issues/802 says we want to use the order defined in boards.txt, while here it looks like we want to follow the one defined in the package index, so I would like to make sure we're doing the right thing before [taking action](https://github.com/arduino/arduino-cli/issues/1834).

per1234, 2022-08-11:
>that the order of the boards defined in the [boards.txt configuration file](https://arduino.github.io/arduino-cli/dev/platform-specification/#boardstxt) is the same as the order defined in the [package index](https://arduino.github.io/arduino-cli/dev/package_index_json-specification/#platforms-definitions). Can you confirm it?

I can deny it. The package author can put any arbitrary content in the `packages[*].platforms[*].boards[]` field. They may happen match if the package author chose to do so, but there is no requirement, and in fact you will often find that the two have different contents.

AlbyIanna, 2022-08-11:
Ok. Can I ask you why in the Board list widget we want to order the boards as defined in the package index, while in the **Tools > Board** menu we want to use the boards.txt one?

per1234, 2022-08-11:
In Boards Manager, we don't necessarily have a `boards.txt` file to get the order from because the `boards.txt` file is only available once the platform is installed, while Boards Manager must list all the available platforms. For non-installed platforms, the only data it has is that provided by the package index.

There really would be no way to use the boards list from package index for the **Tools > Board** menu because that list is not linked to the board definitions other than in an unreliable way by hoping the two human readable strings happen to match (which as I explained they are not required to, and often do not even in the Arduino's primary package index).

AlbyIanna, 2022-08-12:
Understood, thank you for the reply!