arduino/arduino-ide #1117: TEMP on a virtual disk

Andr1961, 2022-06-26:
### Describe the problem

Hi, my TEMP directory is located on a virtual disk in memory , the IDE does not start, writes the error "Error: EISDIR: illegal operation on a directory, realpath 't:\Temp ' ",
can this be fixed? On early RC it started normally. Thanks.

### To reproduce

Imdisk Toolkit utility, create a virtual disk in memory and specify the TEMP folder on it

### Expected behavior

solving the problem

### Arduino IDE version

arduino-ide_2.0.0-rc8_Windows_64bit

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-26:
This issue was also reported by another user on the Arduino forum:

https://forum.arduino.cc/t/arduino-ide-2-rc7-stalls-on-windows-11-system/1002172

They reported there that the regression was introduced at Arduino IDE 2.0.0-rc7 (does not occur when using 2.0.0-rc6).

per1234, 2022-07-21:
In the interest of consolidating all the relevant information in one place, I'm going to close this in favor of another report of the same issue here: https://github.com/arduino/arduino-ide/issues/1236

Even though this one came first, the other has slightly more information. Please do comment over there if anyone has additional information to share.

Thanks so much for reporting this issue @Andr1961

closed 2022-07-21 by per1234

vtsoftware, 2022-09-29:
+1 I'm having a similar problem.