arduino/arduino-ide #1124: Failed to start language server

Wezz19, 2022-06-28:
### Describe the problem

- No splashscreen at start
- Language server fails to start
- The text in the editor has no colors (probably caused by the language server)

The IDE is otherwise working. Verify, upload and serial monitor works normally.

### To reproduce

Place a '#' in the IDE folder name and start the IDE.

### Expected behavior

If '#' in the IDE folder name can't be supported in Windows due to limited support in other OS's then a notification to the user should make it clear that the IDE folder name contains an unsupported symbol.

Otherwise since the IDE seems to be running normally except for the language server then maybe it can be fixed to support '#' symbols in windows at least.

### Arduino IDE version

nightly-20220628

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-06-28:
Thanks for your report @Wezz19. I am able to reproduce the issue when using the Arduino IDE from such a path, but didn't have any problems when running Arduino Language Server directly from the command line with equivalent parameters. So this leads me to believe the issue is here in the Arduino IDE 2.x code base rather than in [the Arduino Language Server code base](https://github.com/arduino/arduino-language-server).

kittaakos, 2022-06-29:
> * The text in the editor has no colors (probably caused by the language server)

No. Syntax coloring is done by [`TextMate`](https://github.com/textmate/textmate) and is unrelated to any language features.

r3inbowari, 2022-09-24:
This is caused by the [`theia-core`](https://github.com/eclipse-theia/theia/blob/1a5bc7a3d8fb07101db45bbe8d12dc6be575cf3d/packages/plugin-ext/src/main/node/resolvers/local-plugin-deployer-resolver.ts#L51) module with incorrect assertion on Windows when the plugins path has a ‘#’.

kittaakos, 2023-01-10:
From the LS logs:

```
{"jsonrpc":"2.0","method":"textDocument/publishDiagnostics","params":{"uri":"file:///private/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino-language-server4280496927/.clangd","diagnostics":[]}}Content-Length: 178

{"jsonrpc":"2.0","method":"textDocument/publishDiagnostics","params":{"uri":"file:///Users/a.kitta/Documents/Arduino/alma%23korte/alma%23korte.ino","version":1,"diagnostics":[]}}Content-Length: 112

{"jsonrpc":"2.0","id":1,"result":null,"error":{"code":-32603,"message":"-32001 Line value is out of range (2)"}}Content-Length: 36
```

This bug might belong to another issue tracker.

kittaakos, 2023-01-10:
I can also see that the URI is incorrectly encoded. Maybe @cmaglie can advise:

```
V[08:48:25.483] ASTWorker running EnumerateTweaks on version 1 of /private/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino-language-server4280496927/sketch/alma#korte.ino.cpp
I[08:48:25.483] --> reply:textDocument/codeAction("2") 86 ms, error: Line value is out of range (2)
V[08:48:25.483] >>> {"error":{"code":-32001,"message":"Line value is out of range (2)"},"id":"2","jsonrpc":"2.0"}

V[08:48:25.484] <<< {"id":"3","jsonrpc":"2.0","method":"textDocument/documentSymbol","params":{"textDocument":{"uri":"file:///private/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino-language-server4280496927/sketch/alma#korte.ino.cpp"}}}

```

Here the URI must be

```
file:///private/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino-language-server4280496927/sketch/alma%23korte.ino.cpp
```

and *NOT*

```
file:///private/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino-language-server4280496927/sketch/alma%23korte.ino.cpp
```

kittaakos, 2023-01-10:
> * The text in the editor has no colors (probably caused by the language server)

I could not reproduce this flaw. The rest is fixed: #1811

@Wezz19, could you please provide the steps to reproduce the syntax coloring problem?

kittaakos, 2023-01-10:
> What I meant with "no colors" is this:

Thanks for the prompt reply. I will check it on Windows. I did not see any issues with the coloring on macOS.

----

> By the way, I think you forgot to edit the last line from your comment 8h ago. You probably meant this:
> 
> > and NOT
> > `file:///private/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino-language-server4280496927/sketch/alma#korte.ino.cpp`
> 
> Not that it probably makes any difference anymore but just wanted to mention it =)

Nice catch. I update it. 👍

kittaakos, 2023-01-11:
I am still seeing errors on Windows.

This correctly works after the go-lsp fix:

```
{"jsonrpc":"2.0","id":1,"method":"textDocument/codeAction","params":{"textDocument":{"uri":"file:///c%3A/Users/kittaakos/Documents/Arduino/min%23max/min%23max.ino"},"range":{"start":{"line":4,"character":2},"end":{"line":4,"character":2}},"context":{"diagnostics":[]}}}Content-Length: 170

{"jsonrpc":"2.0","id":2,"method":"textDocument/documentSymbol","params":{"textDocument":{"uri":"file:///c%3A/Users/kittaakos/Documents/Arduino/min%23max/min%23max.ino"}}}Content-Length: 40

{"jsonrpc":"2.0","id":"1","result":null}
>>>
Content-Length: 184

{"jsonrpc":"2.0","method":"textDocument/publishDiagnostics","params":{"uri":"file:///c%3A/Users/kittaakos/Documents/Arduino/min%23max/min%23max.ino","version":1,"diagnostics":[]}}
<<<
Content-Length: 40

{"jsonrpc":"2.0","id":"2","result":null}
>>>
Content-Length: 143
```

But the Arduino LS fails to publish the diagnostics from `clagnd` to IDE2. I do not yet know what's that, but the URI is not encoded, hence this is a bug:

```
{"jsonrpc":"2.0","method":"textDocument/publishDiagnostics","params":{"uri":"file:///C:/Users/kittaakos/AppData/Local/Temp/arduino-language-server3294049418/.clangd","diagnostics":[]}}Content-Length: 112

{"jsonrpc":"2.0","id":1,"result":null,"error":{"code":-32603,"message":"-32001 Line value is out of range (5)"}}Content-Length: 36
```

----

> I will check it on Windows. I did not see any issues with the coloring on macOS.

I could not reproduce the coloring issue. I tend to split up this issue into two parts:
 - LS is not working if the main sketch file path contains a `#` due to incorrect URL path encoding,
 - syntax coloring is not working on Windows for whatever reason

kittaakos, 2023-01-11:
@Wezz19, could you please help us and try the tester build? I could reproduce the language server issues on both macOS and Windows. It's fixed, but I could not figure out what's causing the missing syntax coloring issue because I could not reproduce it. If you're willing to help, the PR is https://github.com/arduino/arduino-ide/pull/1811, and you can find documentation about the beta testing here: https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md#beta-testing-guide. Thank you!

Wezz19, 2023-01-11:
Okay I tested the zip version on two computers and got the same result on both when I added a # in the path:
[image]
One computer has win10 and the other a freshly installed win11 with no Arduino IDE installed

kittaakos, 2023-01-12:
Thanks for the quick reply, @Wezz19.

I cannot reproduce it. The tester build works fine on my Windows machine when there is a `#` in the sketch path.

Apparently, you have some other issues. The screenshot from https://github.com/arduino/arduino-ide/issues/1124#issuecomment-1379353370 shows there is no `#` in the sketch path unless you have changed your sketchbook location (`directories.user`) too.

I would appreciate some info from you. If you have time, please share with us.

 - Please let me know where you installed IDE2 from the ZIP. What's the full filesystem path?
 - What does IDE2 show when you hover over the editor tab?
 - What does IDE2 show in the browser's console? You can check it on Windows with `Ctrl` + `Alt` + `I` and save the logs.
 - What are in the logs? Please open a `CMD.exe` and start IDE2 with `"Arduino IDE.exe"`, close IDE2, and share the full log.
 - Does the language server works on your side? When selecting any board with an installed platform, do you see syntax errors in the editor?

kittaakos, 2023-01-12:
> I cannot reproduce it.

Now, I can reproduce it. You have a # in the path to the IDE2 and not a # in the path to the sketch. (From here: https://github.com/arduino/arduino-ide/issues/1124#issuecomment-1377546888)

kittaakos, 2023-01-12:
There are two separate issues here:
 - there is a # in the sketch path. It should be fixed by #1811.
 - there is a # in the path to IDE2: https://github.com/eclipse-theia/theia/issues/12064

So this comment 👇 (https://github.com/arduino/arduino-ide/issues/1124#issuecomment-1256816102) is also accurate and valuable. Thank you!

> This is caused by the [`theia-core`](https://github.com/eclipse-theia/theia/blob/1a5bc7a3d8fb07101db45bbe8d12dc6be575cf3d/packages/plugin-ext/src/main/node/resolvers/local-plugin-deployer-resolver.ts#L51) module with incorrect assertion on Windows when the plugins path has a ‘#’.

kittaakos, 2023-01-12:
I could not find any easy way to provide a workaround for this flaw in IDE2. See why: https://github.com/eclipse-theia/theia/issues/12064#issuecomment-1380404909.

> I tend to split up this issue into two parts:

Here is the follow-up issue: https://github.com/arduino/arduino-ide/issues/1815

#1811 fixes the bug when the sketch path contains a `#` and not IDE2 path has the `#`

commit 400f94db (2023-01-11): chore: Use `0.7.4` LS with the URL encoding fixes

Closes #1124
PR #1811 (2023-01-16): chore: Use latest LS with the URL encoding fixes
### Motivation

The Arduino language features must be functional when there is a `#` in the main sketch file name.

### Change description

 - Update the LS version with the fix,
 - Removed an unused module from the code.

### Other information

To verify:
 - Your IDE2 installation contains a `#` in the path. The best is to use something like `C:\Users\You\Desktop\# here is your IDE` because it will ensure that after cutting the `#` (fragment) from the URI, it won't try to load VS Code extensions incorrectly from `C:\Users\You\Desktop\`.
 - You  [... cut]
commit 32e17745 (2023-01-16): chore: Use `0.7.4` LS with the URL encoding fixes

Closes #1124
closed 2023-01-16 by kittaakos