arduino/arduino-ide #1133: Autoformat tab size does not change

tsumanuma, 2022-07-02:
### Describe the problem

Sorry if I'm rude as this is the first time I've used this.

I would like to change the tab size from 2 to 4
I was able to change the width when typing with the keyboard, but not when auto-formatting.

### To reproduce

 In the Command Palette (Ctrl+Shift+P), select "Preferences". Selected "Open Preferences".
I then selected "Editor"; in the Advanced Settings GUI I selected "Tab Size" and changed it.

### Expected behavior

When I press Control + T, the tab is formatted with two sizes

### Arduino IDE version

2.0.0-rc8

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-07-02:
Hi @tsumanuma. Thanks for your report.

As a workaround, I can share the work in progress documentation for customizing the Auto Format configuration according to your preferences:

https://github.com/arduino/docs-content/blob/b43294a6e16e73877e9146ff858c55511e1e0c2a/content/software/ide-v2/tutorials/ide-v2-customize-auto-formatter/content.md

❗ There is currently an error in that document (which will be corrected before it is published on arduino.cc): The file name (`custom_configuration.clang-format`) shown in the screenshot is invalid. The file **must** be named `.clang-format`.

If you would like to change the indent size to 4, copy the configuration provided in the documentation, but change this line:

```yaml
IndentWidth: 2
```

to this:

```yaml
IndentWidth: 4
```

---

In order to make relevant information available to all interested parties, I'll link to the previous report for Arduino IDE 1.x: https://github.com/arduino/Arduino/issues/5012

tsumanuma, 2022-07-02:
> 
It worked!
Thank you so much!

per1234, 2022-07-13:
You are welcome.

I'm glad the workaround was helpful. However, I think that we should eventually make the Arduino IDE 2.x work correctly without the need for such a workaround. So I hope you won't mind that I am reopening this issue so that we can effectively track that task.

I have learned more about the IDE's auto formatter and about the ClangFormat tool recently and I now understand several important things:

## The IDE adjusts the formatter configuration 

Unlike the Arduino IDE 1.x, where the formatter configuration is completely independent from the editor configuration, Arduino IDE 2.x has a sophisticated system for adjusting the formatter configuration according to the user's advanced settings for the editor:

- "**Editor: Insert Spaces**" (`editor.insertSpaces`)
- "**Editor: Tab Size**" (`editor.tabSize`)

The code is here:

https://github.com/arduino/arduino-ide/blob/main/arduino-ide-extension/src/node/clang-formatter.ts

## ClangFormat has a complex way of using tabs

During the formatting process, ClangFormat does the initial indentation using spaces, according to the `IndentWidth` configuration setting value.

If use of tabs for the final indents is indicated by the `UseTab` configuration setting, those spaces are then converted to tabs according to the value of `TabWidth`. For example, if `TabWidth` is set to `4`, then every occurrence of four spaces within the scope of the `UseTab` setting is converted to a tab.

This means there is a complex interaction between the `IndentWidth` and `TabWidth` values, which can produce unexpected results. For example, if `IndentWidth` is set to `2` and `TabWidth` to 4, every two indent levels is converted to a single tab, leaving two spaces for the final indentation of each odd level.

## The IDE does not correctly adjust the formatter configuration

Due to the way ClangFormat uses tabs, the expected output will only be produced when the `IndentWidth` and `TabWidth` ClangFormat configuration settings are set to the same value.

Currently, the IDE adjusts only the `TabWidth` value according to the user's `editor.tabSize` editor setting, leaving `IndentWidth` at the standard value of 2.

So if you have the following IDE configuration:

```json
{
  "editor.insertSpaces": false,
  "editor.tabSize": 4
}
```

The editor will produce the expected results:

[image]

but the IDE's "**Auto Format**" will produce something very odd:

[image]

(the → are tabs and the • are spaces)