arduino/arduino-ide #1135: "Property 'program.tool.serial' is undefined" for arduinoOTA as programmer

JAndrassy, 2022-07-02:
### Describe the problem

I have a programmer in platform.local.txt defined like this:
```
## arduinoOTA as programmer. add entries with {ip} into programmers.txt
tools.arduinoOTA.cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
tools.arduinoOTA.program.params.verbose=
tools.arduinoOTA.program.params.quiet=
tools.arduinoOTA.program.pattern="{cmd}" -address {ip} -port 65280 -username arduino -password password -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b
```
and in programmers.txt I added
```
arduinoOTA115.name=Arduino OTA (192.168.1.115)
arduinoOTA115.program.tool=arduinoOTA
arduinoOTA115.ip=192.168.1.115
```
If I try to upload with this programmer, I get "Property 'program.tool.' is undefined" or "Property 'program.tool.serial' is undefined".
It is the same error for AVR and SAMD platform.

This 'fake' programmer works in IDE 1 and in IDE 2 it works too for board definitions in `hardware` folder ([my_boards](https://github.com/JAndrassy/my_boards))

### To reproduce

for SAMD package create platform.local.txt and modify programmers.txt as described above. then select the "Arduino OTA (192.168.1.115)" programmer in IDE 2 and do "Upload using programmer". The sketch can be Blink and the IP doesn't have to exists.

### Expected behavior

OTA upload should start with 
```
"~.arduino15/packages/arduino/tools/arduinoOTA/1.3.0/bin/arduinoOTA" -address 192.168.1.115 -port 65280 -username arduino -password password -sketch "/tmp/arduino-sketch-E8B0B9BFA9F9DD030AA25850AE73ED4F/Blink.ino.bin" -upload /sketch -b
Connecting to board ...
```

### Arduino IDE version

Version: 2.0.0-rc8 Date: 2022-06-23T09:35:13.756Z CLI Version: 0.24.0 [c1b10f56]  

### Operating system

Linux

### Operating system version

Linux Mint 20.1 Cinnamon. kernel 5.4.0-113-generic  

### Additional context

I received this as issue https://github.com/JAndrassy/ArduinoOTA/issues/161 in ArduinoOTA library repo and I replicated the problem.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-09-26:
Thanks @JAndrassy. I took a look at the documentation for the ArduinoOTA library and identified the necessary update. I submitted a PR for that change: https://github.com/JAndrassy/ArduinoOTA/pull/171

I also surveyed [the other platform configuration files included with the ArduinoOTA library](https://github.com/JAndrassy/ArduinoOTA/tree/master/extras) as well as the platforms in https://github.com/JAndrassy/my_boards and found that no additional changes are needed.

closed 2022-09-26 by per1234

JAndrassy, 2022-09-26:
thank you Per