arduino/arduino-ide #1137: Implement Access Keys (eg menu access via Alt key)

SpitFire-666, 2022-07-03:
### Describe the request

This could be a duplicate of https://github.com/arduino/arduino-ide/issues/588

Currently (on Windows at least), you can't access the menu with the keyboard. Even if you Click on File, you can't use the arrow keys to navigate it.

Would be handy to use Alt+T to open the Tools menu for example.

Cheers

### Describe the current behavior

Currently (on Windows at least), you can't access the menu with the keyboard. Even if you Click on File, you can't use the arrow keys to navigate it.

### Arduino IDE version

Version: 2.0.0-rc8-nightly-20220703 Date: 2022-07-03T03:31:56.192Z CLI Version: 0.24.0 [c1b10f56]  Copyright © 2022 Arduino SA

### Operating system

Windows

### Operating system version

Windows 7

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-07-03:
Hi @SpitFire-666. Thanks for your report.

>you can't access the menu with the keyboard

I am not able to reproduce this on Windows 10.

I can access it using the keyboard via this procedure:

1. Press <kbd>**Alt**</kbd>.
   The "**File**" menu title is highlighted in the menu bar.
1. Press the right arrow key.
   The "**Edit**" menu title is highlighted in the menu bar.
1. Press the down arrow key.
   The "**Edit**" menu opens.
1. Press the down arrow key.
   The first item in the "**Edit**" menu is selected.
1. Press <kbd>**Enter**</kbd>.
   The selected menu item is executed.

Please try that procedure and tell us whether you get the same results I observed.

>Even if you Click on File, you can't use the arrow keys to navigate it.

I am not able to reproduce this on Windows 10.

I can navigate the menu via this procedure:

1. Click "**File**" on the Arduino IDE menu bar.
1. Press the down arrow key.
   The menu opens and the first item in the menu is selected.
1. Press <kbd>**Enter**</kbd>.
   The selected menu item is executed.

Please try that procedure and tell us whether you get the same results I observed.

>Would be handy to use Alt+T to open the Tools menu for example.

We are tracking that deficiency in the dedicated issue here: https://github.com/arduino/arduino-ide/issues/153

per1234, 2022-08-24:
Closing due to lack of response.

@SpitFire-666 if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-08-24 by per1234